EdgeTX/edgetx #2443: 2.8-RC1 - Some telemetry sensor texts selection in the main screen are not translated

robustini, 2022-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I don't know if it's a bug or if these selections are hard-coded, but I don't think so.
I believe instead that they are translated but they remain in English despite having chosen Italian.

### Expected Behavior

They should appear in the chosen language.

### Steps To Reproduce

1. Go to TELE, User Interface, Set Widget
2. Choose a box
3. Click on "select widget"

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

[image]

pfeerick, 2022-10-02:
They are hard coded
i.e. 
https://github.com/EdgeTX/edgetx/blob/823e515392498cdc2e559e1d617b6f564a6a86dc/radio/src/gui/colorlcd/widgets/gauge.cpp#L85
https://github.com/EdgeTX/edgetx/blob/823e515392498cdc2e559e1d617b6f564a6a86dc/radio/src/gui/colorlcd/widgets/value.cpp#L153

Eldenroot, 2022-10-02:
Can we make it into 2.8 final? I can prepare Czech translations...

gagarinlg, 2022-10-02:
> Can we make it into 2.8 final? I can prepare Czech translations...

When you make a PR that adds the necessary defines to all languages and modify the widgets to use the defines, you might or might not get lucky.
But it has to be quick

PR #2454 (2022-10-09): feat: CPP Widget names also translatable
Resolves #2443

All of the built-in widgets now set a display name, which is translatable. The internal name is not changed, hence configured widgets will survive changes in language ;)
closed 2022-10-09 by pfeerick