EdgeTX/edgetx #2444: [SOH]+imagefilename in hash from labels.yml

ParkerEde, 2022-10-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

For two of my models is the hash in labels.yml a little strange. It looks so:
[image]

When I delete the hash and correct the name, the new builded hash will looks like before.
I will attach both labels.yml and model25.yml
[files.zip](https://github.com/EdgeTX/edgetx/files/9692185/files.zip)

### Expected Behavior

the hash should created without [SOH] and image filename in labels.yml

### Steps To Reproduce

copy the attached files to a X10 based radio, cleanup the hash and name in labels.yml, change model several times and power off/on the radio.

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild eff7bfae

ParkerEde, 2022-10-02:
it is also reproducible with the standalone simulator. It is enough to start it only in SD card mode and immediately afterwards these SOH entries are back.

ParkerEde, 2022-10-02:
It seems to be the name itself. When I change it to "Slick Superlit", the problem no longer occurs. The same problem happens when a model is named "TigerMoth 1,86m".
[image]

gagarinlg, 2022-10-02:
So it is the Comma?

pfeerick, 2022-10-02:
I have a feeling feeling 1,86m is a ANSI code... or maybe the syntax is just close enough to screw things up nicely. 

If I load just the model file into simulator for TX16S (i.e. no `labels.yml` at all so it gets generated afresh) I get....

[image]

ASCII 0x01 is SOH, Start of Header... so that still matches what ParkerEde is seeing...

dlktdr, 2022-10-02:
Looks like it's the length of the model name. The value isn't null terminated when it's exactly 15.

[image]

This seems to fix it, will make a PR

gagarinlg, 2022-10-02:
> Looks like it's the length of the model name. The value isn't null terminated when it's exactly 15.

that is a common pratice in the EdgeTX/OpenTX code base to save RAM

dlktdr, 2022-10-02:
See PR# 2448

PR #2448 (2022-10-03): fix: Correct errors in labels.yml if modelname is full length
The buffer for the name doesn't have room for the null if the full length limit.. and used strcpy, so it overwrote extra memory

Removes all strcpy's here with safer strn and terminate the strings

Fixes #2444
closed 2022-10-03 by pfeerick