EdgeTX/edgetx #2451: Intrinsic issues with labels

RC-SOAR, 2022-10-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Labels are at first sight a nice idea, but the way there are implemented could lead to a lot of confusion.  See screenshots below:

[image]
[image]

Each screen shows two models with the same names. 

_Question:_ **how many unique model instances are there?** 

_Answer_: It's impossible to tell know just by looking at the screens. Entries with the same name may be duplicates with different labels, or one instance with multiple labels. 

(In this case the FRSKY model has a single instance with two labels. The FlySky models are duplicates, each with a different label.)

Users will need to take great care when duplicating and labeling models, both of which are common activities.

Folders are IMO a much safe way of categorising, since every model instance must belong to one and only one folder.

### Describe the solution you'd like

Replace labels with folders, 

1 model instance : 1 folder

### Describe alternatives you've considered

An alternative would be to keep labels, but force unique names for every model instance.

gagarinlg, 2022-10-02:
You can also select more than one label and get all models that have both labels.
This is intentional. Probably nobody thought that someone would have multiple models with the same name.

RC-SOAR, 2022-10-02:
> Probably nobody thought that someone would have multiple models with the same name. 

Unfortunately, that happens whenever you select 'duplicate' - the cloned model is given the same name. Many users will just assign a different label, without changing the name. Then start making programming changes to one or other instance. 

A month later, and they may not remember that there are two different instances, and just assume that it's one instance with two labels. There is then the possibility of choosing the wrong instance and crashing.

With a folder/category system as in Ethos and OTX, users know that each entry refers to a unique instance.  There is no confusion, and a reduced possibility of error (no system is perfect!).

If we must have labels, then it would be useful to display the (unique) filename. But my preference is still for the Ethos and OTX   (colour screen version) folder system.

pfeerick, 2022-10-02:
It's not so much an oversight but a user generated confusion that we somehow need to prevent. Do we start altering models when there are duplicates (since the current process is to make a *true* duplicate, so of course the name is the same. Do we enforce unique names (but why should we limit - e.g. I can tell them apart quite easily with different images?). What about showing a indicator, or extra info (like the filename, last modified date) when duplicate model names are detected?

dlktdr, 2022-10-03:
I don't see also how it's that much different than creating two duplicates in a category, but I see where this is coming from. If your not sure how many total instances you have can always uncheck all labels to see everything, then rename as you see fit. Keeping the same name in either case is going to be confusing.

I think forcing unique names could cause a lot of issues/work in code, to many what if's.

I think the idea when creating a duplicate to append a -2, -3 or something to that effect to the name would be a good addition. Then it will would be obvious, and up to the user to properly rename it.

RC-SOAR, 2022-10-03:
> I think the idea when creating a duplicate to append a -2, -3 or something to that effect to the name would be a good addition.

That would work.  Alternatively, display the filename of the highlighted model.

Eiither way, it's essential to be able to identify the root instance easily.

Many users don't have computer skills, having maybe 20 or 30 models, are careless with duplicating and labeling. Expecting them to play around with labels simply to determine which are/are not distinct instances, and which are single instances with multiple labels is not practical IMO.

Eldenroot, 2022-10-03:
I think this has one simple fix - not allow same names for models...every name has to be unique.

dlktdr, 2022-10-04:
Not sure how to handle all cases of the unique name, for instance if I copy a model from one radio to another that has the same name. Which one should be renamed, or once again have a duplicate. 

How about something like this? I don't really want to see the filenames all the time, so use a checkbox to enable/disable them.

[image]
[image]

JimB40, 2022-10-04:
Agree name appendix is better idea. 
If you really need filename add it on top bar in brackets not changing UI layout.
But IMHO then 'after months' user will also not remember that model6 is right and model9 i wrong.

RC-SOAR, 2022-10-04:
I think filename has a major benefit - it's simple and unique. (Name appendix may be re-used after deletions so is not truly unique.)

To save on real estate, and avoid scrolling, the option to display filename could be in System settings.

dlktdr, 2022-10-06:
I've been contemplating this and think what I will work on for 2.9 is have the model name as the filename. Then it has to be unique. Add some logic to rename and duplicate and we get the best of both worlds, no need for extra GUI info and clear what file is what. Will also take into account trying to copy a same filename from one radio to another.

gagarinlg, 2022-10-06:
I am not sure if the file system driver is compiled with UTF support and if windows, MacOS, Linux and the EdgeTX driver will harmonize when using UTF-file names.

dlktdr, 2022-10-06:
Just a thought at the moment. I'm sure there will be lots of other issues that could arise and make this not possible as well.

RC-SOAR, 2022-10-06:
> I've been contemplating this and think what I will work on for 2.9 is have the model name as the filename. Then it has to be unique. Add some logic to rename and duplicate and we get the best of [...]

I can't see how this will work. Model names are... descriptive labels. Filenames are properties of the file system. 

Why not blatantly copy from a system that is known to work. In Ethos, when you highlight a model, it displays the filename and last updated date. In tiny font, so tiny you almost have to use a magnifying glass. So it takes up little screen real estate. 

Best to keep it simple.

RC-SOAR, 2022-10-16:
Related issue: 

When opening the Model Select menu, I have no idea what the label(s) for each model. I can only determine the labels by opening the label editor, or the model setup menu. 

When I go to the Model Select menu, I want to know exactly what I'm opening _now_, without fiddling with other menus.

The solution would be to show all the labels along with each model, but of course there is the lack of screen real estate.

The more I try to use labels with my own models, the more I feel that they're a complicated solution in search of a real problem. Sorry.