EdgeTX/edgetx #2462: 2.8-RC - Bluetooth Trainer practically unusable

robustini, 2022-10-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I had already reported this problem in OpenTX at this time, and obviously in EdgeTX we brought it back.
Yesterday I tried again between a Tandem X20S with ETHOS firmware and my X10S Express with "modded ISRM" for the "MPM Trainer" option enabled.
In BT trainer mode the commands given by the slave's tx are frighteningly jerky, if instead I use MPM as RX they seem to be given by the master tx, very fluid.
So I tried again between two X10S Express and the result is the same, all jerky.
ETHOS I saw that in the bluetooth options it has a so-called "high speed" mode which significantly increases the range between the two tx, but the problem remains the same, not having two Tandem here I don't know how it would behave.
I can't believe it's a bandwidth issue as Bluetooth has some available so I don't understand where this limit is.

### Expected Behavior

The commands given by the slave tx should be linear and not jerky.

### Steps To Reproduce

1. Connect two tx in bluetooth trainer
2. Check the linearity of the movement of the servos
3. Do the same thing using two MPMs in "Multi Trainer" and you will notice the difference

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

robustini, 2022-10-03:
Even in video you can see the difference very well.
Here the slave commands using MPM:

https://youtu.be/DdHq3StWEiY

And this instead using Bluetooth:

https://youtu.be/SNyVtF1gI3E

robustini, 2022-10-03:
> @robustini are you saying this is the same with Ethos?

I have not tried with two TXs with ETHOS, but only between EdgeTX<->EdgeTX and ETHOS<->EdgeTX.

raphaelcoeffic, 2022-10-03:
@robustini I’m afraid it’s always been this way because the module firmware is basically broken (by design?).

robustini, 2022-10-03:
@raphaelcoeffic so the only ones who could get into it are those of FrSky, who don't give a damn, great!
So the bluetooth trainer is useless.

robustini, 2022-10-03:
But something they did in the module for the Tandem X20 because it has a few more options.
I don't understand why they left out the PARA module on the X10S Express and the other on the X12S, maybe @bsongis he knows more than we do about that, since he developed ETHOS.

dlktdr, 2022-10-04:
> ETHOS I saw that in the bluetooth options it has a so-called "high speed" mode

I'm guessing this is enabling CODED PHY Bluetooth mode, which has a much longer range and is more reliable. If they haven't made new modules and the code on the PARA ones is the same as previously produced, If you could capture the RX line into the module and see what they are sending to make this happen. The chip used should support this mode CC2650 (as best we could tell) I have tried connecting to the module and requesting this mode, but it won't switch away from PHY 1M.

What other options do they list on ethos?

robustini, 2022-10-04:
@dlktdr thanks Cliff for your detailed explanations.
The Tandem X20S is from a friend, I don't have it here but I'm sure of that selectable "high speed" option.
In my opinion they have changed the type of module, something perhaps more performing, and what is quite interesting even more than the trainer is the "audio" option available only on high-end tx, this proves that there is a different hardware.

[image]

robustini, 2022-10-04:
This is the Tandem X20S, the module looks like the usual PARA.

[image]