EdgeTX/edgetx #2463: Naming switches on hardware screen causes issues in inputs & mixer screens

EhAye, 2022-10-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After naming the SA switch in the Hardware screen to anything, the icon used for the SA switch is changed to what looks like a trim switch in some places

The switch's icon is replaced with a "-" on an Input's "Source" dropdown, and hides the name of the input source from the Inputs screen

### Expected Behavior

renaming the hardware switch should not change the switch's icon, and should not hide the source on the Inputs screen

### Steps To Reproduce

1. Press SYS button, and select Hardware screen
2. Rename SA switch to "ANY"  (can be anything)
3. Return to main screen
4. Press MDL button, select Inputs screen
5. Select Input row #5, click Edit
6. Click Source dropdown box
**BUG**: You'll notice the SA/"ANY" switch icon has changed, along with some other switches.

7. Select the SA/"ANY" switch as the input
BUG: You'll notice the selected source is named "-ANY" without any icon, and with a dash preceding the name.

8. Press Back button to return to main Inputs screen
**BUG:** You'll notice the source for input row 5 is not listed... You'll see "100%" for the weight. but the input is hidden.

9. Return to main screen
10. Press SYS button and select Hardware screen
11. Remove the name from SA switch
12. Return to main screen
13. Press MDL button, select Inputs screen
**BUG:** You'll notice the input is no longer hidden.

Finally, tap Input row 5, click Edit
The dropdown for "Source" shows correct switch icon, everything back to normal.

[image]
[image]
[image]
[image]

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

happens in the simulator too

gagarinlg, 2022-10-03:
I just checked with 2.8 RC2, there it is a little bit less bad. The Icon is still gone, but at least the name is shown everywhere.

gagarinlg, 2022-10-03:
So, after reading the code, the icon is part of the default switch name (and not only switches, it is the same for other sources). If you change the name, you loose the icon.

pfeerick, 2022-10-09:
As Malte commented, 2.8 doesn't exhibit the weird icon behaviour, but you won't get a switch icon. This might be something we should investigate in 2.9, as it seems a little odd that the other inputs do keep their icon if they are named.  But the bugged behaviour is gone now at least. 
[image]
[image]
[image]

closed 2022-10-09 by pfeerick