EdgeTX/edgetx #2466: add single expo function in curve menu on radio

MicheleVilla, 2022-10-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In companion curve menu is possible to generate a single expo curve, on radio this option is missing
single expo is very useful, for example, to regulate the gas of internal combustion engines

### Describe the solution you'd like

add the single expo option in curve menu
it would be very useful if the coefficient of the curve could be modified by associating it with a global variable

RC-SOAR, 2022-10-05:
Here's a way to turn a 2-sided expo curve to 1-sided for Motor, also adjustable in the air via a GV: 

https://rc-soar.com/opentx/varthrcurve/index.htm

Eldenroot, 2022-10-06:
@pfeerick I think this should not be closed but marked as feature request... makes sense!

MicheleVilla, 2022-10-08:
I reopen the issue
if you think there cannot be a solution, close it

Eldenroot, 2022-10-12:
You mean something like this?

[image]

https://youtu.be/I5dCLQW54ps?t=347

EXPO Rate A and B ?

raphaelcoeffic, 2022-10-13:
More like this:

[image]

MicheleVilla, 2022-10-13:
**My original request is like this**

> More like this:
> 
> [image]
My original request is this,  

**This solution is very intersting**

> You mean something like this?
> 
> [image]
> 

> https://youtu.be/I5dCLQW54ps?t=347
> 
> EXPO Rate A and B ?

This curve management affects the inputs menu, but can it also be implemented in the curves menu at the same way?

MicheleVilla, 2023-04-07:
Is there any possibility that this request will be developed?

MicheleVilla, 2024-02-12:
I am using the solution proposed by RC-SOAR to manage the single exponential curve on the throttle control, but I think that being able to apply it natively would be easier to manage and its display more understandable in the diagram on the input page.
I also reiterate that the solution of managing the two parts of the curve separately as proposed by Eldenroot is very interesting.
Of course if there are problems implementing this solution or it is considered not interesting for the development of EdgetTx I can close my request without problems
Thank you