EdgeTX/edgetx #2478: 2.8 Output Editor - Highlight Weight for Stick Movement

jmxp69, 2022-10-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When editing an output the Min Max value should highlight according to stick movement.

i.e. If I move the stick right the Max label should be bold or inverted or somehow colorized to indicate the output weight that should be adjusted corresponding to that stick movement. On the outputs page this behavior is evident. Stick movement causes a weight to highlight in bold. 

### Describe the solution you'd like

In the output editor, the weight corresponding to stick movement should be highlighted like it does on the outputs page.

### Describe alternatives you've considered

This feature exists in 2.7.1. It appears to be a regression.

gagarinlg, 2022-10-04:
[image]

PR is made
Maybe it is enough to highlight the label and not the edit field and the button?

jmxp69, 2022-10-04:
Yeah, any highlight would be sufficient. Definitely take the path of least resistance.

Eldenroot, 2022-10-05:
Hmm, not visible at all (at least for people wearing glasses/outside during sunny day)... there should be used different background color + bold text...

gagarinlg, 2022-10-05:
like this:
[image]

Eldenroot, 2022-10-05:
Yes, just maybe use different background color (not same as selection color) and make also the value bold

gagarinlg, 2022-10-05:
There is only so many different color options available...
[image]

Eldenroot, 2022-10-05:
Coool

PR #2480 (2022-10-06): fix(gui): add missing highlight of the Min/Max value in output editor…
… acccording to the input

Fixes #2478 

Summary of changes:
highlights the label and edit fields of the Min/Max values in the output editor

This PR needs EdgeTX/libopenui#73
closed 2022-10-06 by pfeerick