EdgeTX/edgetx #2479: 2.8 Curve Page Touch Area Not Intuitive

jmxp69, 2022-10-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

On 2.8 Curve page there are some consistency issues.

1) Clicking Curve Numbers

- If no curve exists, I can click on a curve number and a pop-up offers Edit and Preset -- Ok
- If a curve exists, I cannot click on the curve number to activate the pop-up offering Edit/Preset/Mirror/Clear. 

This is inconsistent. If we can click a number to edit a blank curve, we should be able to click a number to edit an existing curve.

2) Clicking Curve or Whitespace

- When a curve exists, pressing the graph with the curve in it yields haptic feedback indicating from radio, "yes, I register your finger press" but the edit dialog box does not appear.
- When a curve exists, the user must press on the whitespace to the right of the graph.

This feels un-intuitive as all other screens allow pressing anywhere on the control to bring up the edit dialog option. 

### Describe the solution you'd like

I would like to not be a monkey with a football.

EdgeTX team can help by doing one thing:
Extend the finger-press capture region for curves across the entire curve line from the curve number to the far right of the screen unless there is a reason not to...?

### Describe alternatives you've considered

Relaunching my amphetamine habit.

pfeerick, 2022-10-06:
I can't see any reason to not make it whole line- other than the curves page basically hasn't been touched at all yet :) Some of this is pre-2.8 behaviour plus some regression due to making it work again at all. So will most likely be a 2.9 change as more pages get updated/refreshed.

jmxp69, 2022-10-07:
Yeah, not a show stopper. Just found this and had the lightbulb click on while going through 2.8 front to back.

gagarinlg, 2023-02-26:
Can you test the current nightlies?
I think this is not an issue anymore

closed 2023-04-07 by pfeerick