EdgeTX/edgetx #2481: Chanel monitor error!

Airfiddler, 2022-10-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If a servo is reversed ie: Throttle servo, the channel monitor still shows movement as if the servo is not reversed. If my throttle is -100 I would expect the monitor to reflect that, by lighting the left side of the monitor. Currently the 2.8 rc2 release lights up the right side of the monitor at low throttle, even with a reversed servo. Quite confusing if you didn't remember the Throttle servo was reversed.

Excellent work on this update.

### Expected Behavior

Throttle stick low ( with a reversed servo) = -100 or left side of channel monitor lit up.

### Steps To Reproduce

1: Reverse throttle servo.
2: Observe channel monitor.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

October 3rd nightly build

eshifri, 2022-10-05:
"Reversed servo" means reversed output?
Please notice that channel and output are not the same.

Airfiddler, 2022-10-06:
> So are you specifically talking about the Throttle -> Reverse option? As
> that does not reverse the servo - it reverses the persitoin the throttle
> stick is expected to be in... so everything is working as it should be, and
> this is not a bug ;)
>
> A check box in Model Set Up to ensure correct operation of throttle-based
> functions and timers for set ups in which full throttle is in the down
> position. CAUTIONS: This check box does not reverse the direction of the
> throttle servo nor does it reverse rotation of the motor.

Thank you for your reply.  I believe as you say, everything is working
properly. My confusion was in the channel output monitor, when reverse
throttle (CH 3)was selected. My apologies. You guy's are doing amazing
work. Please close this issue. Thanks again for your help.

pfeerick, 2022-10-06:
It's always good to do a sanity check... Some of the options like that one are "strange" since they are there for specific purposes. :)

closed 2022-10-06 by pfeerick