EdgeTX/edgetx #2484: 2.8 RC1/2 ExpressLRS ETX Passthrough flashing fails

pkendall64, 2022-10-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

After upgrading from 2.7.1 to 2.8.0-RC1 or 2.8.0-RC2 ETX passthrough flashing of the TX16S internal module fails.
If I downgrade back to 2.7.1 then passthrough flashing works again.

### Expected Behavior

Passthrough flashing should continue to work after upgrading.

### Steps To Reproduce

Use the ExpressLRS configurator and select 3.0.0, Radiomaster tx16S MKII, EdgeTxPassthrough flashing it will detect the chip, but fail after uploading the stub loader.
Interestingly using my online flasher (https://pkendall64.github.io/elrs-web-flasher/) it works because it uses a slower baud rate for updating.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-10-06:
> As an aside: I have noticed that the ExpressLRS Lua script progress bar is quite a bit slower in 2.8 vs 2.7.1

i believe this is related to the screen refresh rate. Possibly more infos could polled between screens.

gagarinlg, 2023-05-05:
Can anyone check if this is still an issue with the current nightlies?

pkendall64, 2023-06-25:
Yes this is actually fixed. Although with 2.9 we have a different issue, which I will open a ticket.

closed 2023-06-25 by pkendall64