EdgeTX/edgetx #2487: Update Companion 2.8.0

Charlevia, 2022-10-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Access denied.

### Expected Behavior

Update

### Steps To Reproduce

Request update from Companion 2.8.0
Ubuntu latest.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]

elecpower, 2022-10-08:
@Charlevia are you using RC or self build?

Charlevia, 2022-10-08:
RC?   I am old and have trouble with acronyms.
RC then, self build is not on my list.

gagarinlg, 2022-10-08:
> RC? I am old and have trouble with acronyms. RC then, self build is not on my list.

RC = Release Candidate

closed 2022-10-14 by Charlevia