EdgeTX/edgetx #2495: Screen display doesn't work/missing on commando 8

CcLaFleche, 2022-10-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I can adjust the values to display, voltage, consumption, etc...
But cannot be displayed.

### Expected Behavior

nothing else

### Steps To Reproduce

nothing else

### Version

2.7.1

### Transmitter

iFlight Commando8

### Anything else?

lua script for BF doesn't work too, i think....
Impossible to change my PID on radio or access to anything.

pfeerick, 2022-10-09:
So by "can adjust the values to display, voltage, consumption, etc..." do you mean setting up the telemetry display/screen (last page of model settings)? If so, you show that page with by pushing the right joystick to the left for about a second (joystick, not the gimbal)

With the Betaflight Lua, are you saving the settings? As if you only change them and don't save, they are only changed on the screen. To save/send to the flight controller, you long press the right joystick/long ENT, and choose the 'save' opt to send all the settings for that page. 

Both are working for me with iFlight Commando8 and BetaFPV Meteor75.

CcLaFleche, 2022-10-09:
Thanks !
For the display, you're right, I make myself Coptic that I simply did not know the manipulation.
Where can I find a complete and detailed explanation of this radio?
I can't find much on her....

For the lua script it shows me error syntax, maybe I don't have the right version on the SD?

pfeerick, 2022-10-09:
> Where can I find a complete and detailed explanation of this radio?

Basically nowhere... it's still too new for there to be any real information on it. I did try to annotate the pull request adding support with some of the more unique quirks or behaviours of this radio. But I don't remember mentioning anything about that. 

> For the lua script it shows me error syntax, maybe I don't have the right version on the SD?

That is possible - sometimes you'll get an error the very first time some of the larger Lua scripts run, and then it's fine from them onwards. Either way, probably a good idea to get the current version from https://github.com/betaflight/betaflight-tx-lua-scripts/releases just to be sure.

CcLaFleche, 2022-10-09:
Thanks a lot for your time !
All fonctions work fine !!!
Thanks !!!
You can delete my post I think... 😅

pfeerick, 2022-10-10:
Nah, it'll be useful for someone else I'm sure... ;) Glad it's working fo you. Don't hesitate to jump on our discord (you'll find the badge/link on the front page of the repo) if you get stuck with anything.

closed 2022-10-10 by pfeerick

CcLaFleche, 2022-10-10:
Well thank you for all that, it changes so much from the reception in other working groups that I won't mention...
🙏🙏🙏