EdgeTX/edgetx #2498: FM Global Variables

CarlKnox, 2022-10-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

v2.7 would list the GV1-9 in each FM so you could label them or see how they change as switches are thrown.
v2.8 rc2 does not show GV.
I have widgets using GV, so I know they are still in use and working. I just can't see them from the Tx.

### Expected Behavior

Add GV to FMs similar to Trims, perhaps.

### Steps To Reproduce

Press Model button
Select Flight Mode
Select FM0
[image]

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

v2.8.0-rc2

pfeerick, 2022-10-09:
Are you sure you're not confusing the FM tab with the GVs tab? This is the FM tab from 2.7...
[image]
... and this is the GV tab (which is exactly the same in 2.8 as it hasn't been touched... yet!)...
[image]

CarlKnox, 2022-10-10:
Maybe move this to a feature.
I'm thinking if GV can see (monitor) FM, then FM should also see GV.

pfeerick, 2022-10-10:
I guess then really need to think about:
- what is the purpose/benefit of showing it in the Flight Modes screen (as opposed to pre-existing GV screen)?
- how would it look?
- how would it work?

raphaelcoeffic, 2022-10-10:
@pfeerick A friend of mine kept bugging me with this as well. The point is: you can witness it on the `128x64` UI (and only there!).

I think basically having a table with the GVs and values.

pfeerick, 2022-10-11:
lol... ah, ok, so the 128x64 doesn't have a GV page at all... that would explain it 🤣 

Interesting... so people actually want the regression in capability of the 128x64 B&W UI (since the 212x64 B&W UI also shows like colorlcd)? Where you have to go into a flight mode to see the GVs? As opposed to being able to see all the GVs at once and which flight mode is using them? I bet not... pretty sure they want it both ways, and the cake too! 🤣 So that's what they'll get (not the cake, I'm keeping that!).

raphaelcoeffic, 2022-10-11:
Sometimes you might want to set the GV for a single flight mode, thus editing more than one at the same time, and sometimes you might prefer to edit a GV for all flight modes at the same time. Really depends on what you’re trying to achieve.