EdgeTX/edgetx #2499: 2.8 RC2 - Copying and pasting anything in the tx menu when complete the selection go back to the top

robustini, 2022-10-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Is normal that when copying and pasting anything in the tx menu, for example in the special function page page, when i exit the selection go back to the top of the list?
For example, I copy and paste a special function SF12 to SF13, once the selection is made it goes back to SF1, while in my opinion it should stay on SF13.
It would be convenient for it to remain where I did the operation, so it's really boring.
Is it a bug or is it intentional?

### Expected Behavior

After any operation it should remain in the target one.

### Steps To Reproduce

1. Copy and paste a special function

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

pfeerick, 2022-10-09:
And just to make sure there is no confusion, this is the *selection* / cursor, and is a problem particularly with non-touch, as it means you need to scroll back to where you were before after a paste/insert/clear/delete, etc.

robustini, 2022-10-09:
Yes, the problem is evident with non-touch tx.

raphaelcoeffic, 2022-10-10:
@robustini as far as I can see, it happens only with the "old" screens. With inputs/mixers, it returns to the same spot. However, I'm not sure why the pasted element is edited directly. I believe this is not intended. Regarding the "old" screens, I think this should go away once these screens are redone with LVGL flex layout.

pfeerick, 2023-03-16:
Fixed with LVGL version of these pages, and some other optimisations.

closed 2023-03-16 by pfeerick