EdgeTX/edgetx #2500: Warnings require touch screen to skip on TX16S

CamGenius, 2022-10-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Turning of the radio before the receiver gives you the warning about the receiver still being connected. The only way to bypass this is to touch the touch screen skip button.

### Expected Behavior

Pressing the enter button on the roller should also press the skip button

### Steps To Reproduce

1. Turn on TX and RX with telemetry enabled
2. Once connected turn off TX
3. Try to skip warning with enter button on the roller wheel

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.8.0-RC2

pfeerick, 2022-10-09:
Aw man, not again! This one keeps coming back (or was it never fixed... hm...) ... Then again, there have been a few PRs over the last few versions of ETX altering stuff around this area, so it could have been knobbled by one of them... this one springs to mind, but will need further investigation.

Hm, more detail needed I think... as I can't repo this (I am on a later dev build than RC2, but I don't *think* anything changed that would effect this). So what protocol is this with - i.e. is this the MPM? As I was just testing with CRSF/ELRS module that was closer at hand to have something with telemetry... and both touch and roller were confirming power off. And this *is* the shutdown / model still powered  screen that has [ENTER] and RTN text at the bottom?

eshifri, 2022-10-09:
I can reproduce it with MPM and AFHDS2A

pfeerick, 2022-10-10:
Yeah, I've not looked at the code, but I was also able to trigger it when going from Frsky D16 => D8 models (spoiler, D8 doesn't have a TX programmable failsafe setting) and since it seemed delayed, I inclined to think something is no longer being handled right with the delayed MPM failsafe check.

CamGenius, 2022-10-10:
It works as expected with elrs but with mpm it doesn't work right for me.

pfeerick, 2022-10-10:
Hm... ok... what did I miss... or is this actually working for me for a change, and not for you? 😆 

https://user-images.githubusercontent.com/5500713/194795051-9abfc711-f7c6-4d1f-9ddd-71426608dada.mp4

eshifri, 2022-10-10:
I have used my private build with the current main code...

pfeerick, 2022-10-10:
That's neither here nor there... I've been doing the same, and ended up going back to the same build stated as being used, but still don't seem to be able to reproduce this. So am trying to find out if there is something more to this.

eshifri, 2022-10-10:
Got it: a very short press on the roller switches the TX off. If you press longer (anything that can be interpreted as a LONG_PRESS) - no reaction. I think normal press also does not work - it should be shorter than normal, but I may be wrong.

pfeerick, 2022-10-10:
Ye ye ye... You get the gold star today! 🥇 Yes, a "normal" and long press are basically unresponsive... Well, that's just annoying... when is an ENT not an ENT? 🤣

eshifri, 2022-10-10:
Looks like all events except LV_EVENT_SHORT_CLICKED are intercepted and removed somewhere...

pfeerick, 2022-10-10:
Ok, so the underlying issue for that particular one _might_ actually be because the ui is being run manually since power checks are ignored... `run_ui_manually()` in in effect

However, this also effects other dialogs, i.e. the throttle warning on startup... oh, interesting... long RTN was handled in that case, but not long ENT

eshifri, 2022-10-10:
https://github.com/EdgeTX/edgetx/blob/af45201f64cd0c2d10b201b4706b26a1f95e975a/radio/src/opentx.cpp#L765
SetCloseCondition() is based exclusively on timer.

raphaelcoeffic, 2022-10-10:
Gentlemen: I just tested this, and it seems to me to be working properly:
- press `ENTER`, and it will shut down.
- press `RTN`, and it will cancel.

Is that really about accepting `[long ENTER]` as well?

pfeerick, 2022-10-10:
Yes, that is what it boils down to generally for any fullscreen dialog... so that a longer than brief ENT is also accepted. Doesn't necessarily have to be on press, can be on break.

raphaelcoeffic, 2022-10-10:
@pfeerick that looks like a feature to me, not a bug. It is working as designed.

commit bece3401 (2022-10-11): feat: FullscreenDialog also support long pressed ENTER (#2506)

Resolves #2500
commit 45632535 (2022-10-12): feat: FullscreenDialog also support long pressed ENTER (#2506)

Resolves #2500
closed 2022-10-11 by pfeerick