EdgeTX/edgetx #2508: The pointer is the position of the stick on the curve graph setting.

Alex6892, 2022-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There is no stick position indicator on the chart when setting the curves function.

### Describe the solution you'd like

Is it possible to get a pointer to the stick position on a curve graph setting, similar to the graph in the input menu?
It is very useful to see the position of the stick on the chart when adjusting the curves function.

### Additional context

RM  TX16S  2.8.0-RC2

Eldenroot, 2022-10-10:
I miss this feature from Spektrum radios... it is really useful to see the actual value and check the curve with stick...

Alex6892, 2022-10-10:
Thanks, if there are many of us, the developers can make a change.

MicheleVilla, 2022-10-10:
in the curve display menu a solution could be to choose which source to use to move the cursor along the x axis

raphaelcoeffic, 2022-10-10:
> in the curve display menu a solution could be to choose which source to use to move the cursor along the x axis

Yeah , this is basically the main blocker: which control should be used to display the curve movement? Then, should this control still work normally while used to show the curve values? Frankly, I have only bad solutions or cumbersome to implement...

gagarinlg, 2022-10-10:
Maybe a slider on the touch screen, also usable with the scroll wheel of course, would be a good compromise, instead of using a stick.

Eldenroot, 2022-10-10:
Curve screen need some love :)

raphaelcoeffic, 2022-10-10:
**Crazy idea:** what about using **auto-source** to select which control is used?

Imagine: you move just any control (ok, not just any, more like knob/stick) and it starts moving the cursor from -100 to 100.

Alex6892, 2022-10-10:
In the menu (inputs) everything works fine, even shows the coordinates.
You need to do the same in the menu (curves).
I can't program.

gagarinlg, 2022-10-10:
> **Crazy idea:** what about using **auto-source** to select which control is used?
> 
> Imagine: you move just any control (ok, not just any, more like knob/stick) and it starts moving the cursor from `-100` to `100`. Of course, that would not change anything from the what the control actually does.

Great idead

> In the menu (inputs) everything works fine, even shows the coordinates. You need to do the same in the menu (curves). I can't program.

That is clear. The discussion is about what input to use to move the indicator, because curves do not have a specific input assigned to them a decision has to be made what input should it be

MicheleVilla, 2022-10-10:
imagine adjusting the curve associated with the stick of the internal combustion engine or the compensation of the spoilers in this case you could adjust the points of the curve in real time
in any case the source could also be another channel in this case the excursion would no longer be +/- 100
A drop-down box could summarize all possible sources

gagarinlg, 2022-10-10:
What @raphaelcoeffic proposed was to use the last input that has been moved as the indicator source.

Eldenroot, 2022-10-10:
Guys you are awesome, this brainstorming is the hearth of this EdgeTX community! I really appreciate it, not only from you devs but also from regular users!

MicheleVilla, 2022-10-10:
thanks to the developers for their patience and helpfulness !
in Input and mixer menu a long press on name of  selected curve (if cstm option is selected) displays the curve menu with the input (or mixer) from which you started as source
this does not happen in the output menu
can it be a solution to what we are looking for?

pfeerick, 2022-10-11:
I think the underlying problem is still that curves aren't tied to a specific input/mix/output (since they can be used in all three places). So what happens when more than one input/mix/output uses them? This is when auto source/control selection on the curve page would make sense.

Alex6892, 2022-10-11:
Where is the curve mixed in on the block diagram?
Where to see the picture?

eshifri, 2022-10-11:
> Where is the curve mixed in on the block diagram? Where to see the picture?

There is a separate TAB for curves. The curve is not associated with anything (input/mixer) there - it is a "curve in itself".

MicheleVilla, 2022-10-11:
> I think the underlying problem is still that curves aren't tied to a specific input/mix/output (since they can be used in all three places). So what happens when more than one input/mix/output uses them? This is when auto source/control selection on the curve page would make sense.

I a curve is open by a long press in menu input and mixer the cursor is associated with these
for example mixer ch1 is associated at Input 1 which is associated at thr stick
In mixer 1 is selected CV1
visualizing curve 1 by long press on CV1 in mixer 1 the cursor is moved by thr
same behavior in Input menu
in the output menu there is a strange behavior of which I cannot understand the logic

eshifri, 2022-10-11:
A curve editor can be opened independently of any input or mixer as well.

MicheleVilla, 2022-10-11:
generally a curve is used to modify what enters and that is associated with the X axis with a series of points that represent what comes out and that is represented by the Y axis
The curve can be used in several menus
Mixer inputs outputs
here the useful thing is to make the cursor move from the data processed by the function from which it was entered (input mixer output) in such a way as to always know which point of the curve to modify
Naturally, if the curve is modified from the CURVE menu, a cursor is not displayed, but the points are only modified via the keyboard
In this case it could be useful to be able to select a source or a channel that is associated with a cursor that moves along the X axis
Perhaps this type of function is not what the author of the discussion required (the translation is probably a bit misleading) but the discussion is very interesting anyway

Eldenroot, 2022-10-11:
Please check this short video - I miss this from Spektrum TX.

It was so easy to set throttle/pitch or whatever curve you wanted

https://youtu.be/kpRKVUsDDwk?t=83

raphaelcoeffic, 2022-10-11:
We really need to improve these curves, that is clear 😇

Alex6892, 2022-10-11:
> таким образом, чтобы всегда знать, какую точку кривой изменить.

This is the main purpose of the cursor in my desire.

Alex6892, 2022-10-11:
> here the useful thing is to make the cursor move from the data processed by the function from which it was entered (input mixer output) in such a way as to always know which point of the curve to [...]

This is the main purpose of the cursor in my desire.

MicheleVilla, 2022-10-11:
"here the useful thing is to make the cursor move from the data processed by the function from which it was entered (input mixer output) in such a way as to always know which point of the curve to modify"
I add that is already possible in menu Input, Mixer and output by a long press on selected Curve name in option CURVE->cstm ->CVx
My radios are 9XD and 9XE

pfeerick, 2022-10-11:
Just a reminder, the OP is talking about colorlcd radios ;) So would need to check if the wide-screen B&W UI exhibits the behaviour desired for colorlcd, and catch it up when the curves screens get their refresh. Since the implementations are completely different for B&W and colorlcd.

Alex6892, 2022-10-11:
> I add that is already possible in menu Input, Mixer and output by a long press on selected Curve name in option CURVE->cstm ->CVx
> My radios are 9XD and 9XE

It is, except for (output), but the cursor is nowhere to be found.
My radio RM TX16S 2.8.0-RC2

pfeerick, 2022-10-12:
No, he means on the 212x64 B&W radios (but not the 128x64 i.e. zorro, tx12, tlite etc) it actually does this... so colorlcd and 128x64 B&W need to catch up ;)

https://user-images.githubusercontent.com/5500713/195221795-ca280fdb-1243-4971-a5fc-267d115de5f1.mp4

eshifri, 2022-10-12:
Interesting: in OTX 2.3.14 it does not work this way.
ETX got something new for Taranis, but not for color?

pfeerick, 2022-10-12:
c'mon, you know better than that @eshifri 😆212x64 doesn't get any more attention than 128x64 on ETX... so if we had added it (spoiler: OTX doesn't have it on 128x64 either) then both B&W resolutions would have got it. It's definitely present on OTX 2.3.15, and I doubt it's a recent addition. 

Instead, it will be yet another feature that wasn't finished in the OpenTX 2.4 branch... since OTX 2.3 does have this on colorlcd:

https://user-images.githubusercontent.com/5500713/195233104-3072ac1b-61af-43c4-a4ed-9933747c2c2d.mp4

eshifri, 2022-10-12:
Actually, it works in OTX, but not for any curve: it works only if you go to curve edit from a mixer (or, maybe, input as well).
If you got to curve edit from the Curves page - no cursor.

gagarinlg, 2022-10-12:
EdgeTX has been build from the OpenTX 2.4 dev branch.
That was never completed and some functionalities were not implemented, that exist in OpenTX 2.3. So anything missing in EdgeTX, that exists in OpenTX 2.3 was already missing in the OpenTX 2.4 branch. Some stuff has been added now in EdgeTX and now the UI will start to go even further away from OpenTX 2.3 and OpenTX 2.4

pfeerick, 2022-10-12:
There is no list of missing features, because, like with this issue, we are still finding them as people spot them. 🤣You are welcome to sit down with OpenTX and EdgeTX and compile a list for us ;)

closed 2022-12-06 by pfeerick

Alex6892, 2022-12-07:
> Списка недостающих функций нет, потому что, как и в случае с этой проблемой, мы все еще находим их по мере того, как люди их замечают.рофлВы можете сесть за стол с OpenTX и EdgeTX и составить для нас список ;)

Thanks for the work done https://github.com/EdgeTX/edgetx/issues/2508 , How can I use it, there is a ready-made nightly firmware for RM TX16S?

PR #2580 (2022-12-06): feat(color): Display current stick position on CurveEdit screen

Fixes #2508 

This PR adds functionality to display stick position on the edit curve screen.
Functionality is similar to OTX and ETX B/W: if you enter CurveEdit from input or mixer the stick position is shown; if you go to this screen from CURVES tab - no cursor is shown on the curve.
(It is possible to keep "last used stick", but it does not look right to me.)