EdgeTX/edgetx #2512: Number entry regression

ducatiJake, 2022-10-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In prior versions of Edge, when needing to change Weight or offset or any other number, the bottom of the screen would be replaced with the 'MIN DEF +/- MAX' entry controls and the jog dial would still increment or decrement the value I was trying to change.  It used acceleration to help move a large amount.  I found this the easiest way to change a number.  Never found those controls at the bottom very useful.  In 2.8 the jog dial now just moves the highlight on the buttons at the bottom of the page.  Very awkward now.  Please revert to the 2.7.1 behavior and let the jog dial change the number being entered. 

### Describe the solution you'd like

Please revert to the 2.7.1 behavior and let the jog dial change the number being entered. 

### Describe alternatives you've considered

I would keep the old jog dial behavior in any case, but how about just a simple numeric keypad entry control just the numbers 0-9. Probably need a delete or backspace button.

eshifri, 2022-10-11:
You are mixing touch and button interfaces together.
If you use the roller to select the field, press ENT and roll the wheel you will get the behavior you are asking for.
If you use touch interface to select the filed, the keyboard opens and the roller moves across the keyboard fields.
So the situation is not as bad as you describe. :-)

pfeerick, 2022-10-11:
As eshifri said, this only happens if you use touch to select the field, and *then* move to the rotary encoder. If you select/press ENT on the field with the rotary encoder, it controls the field directly, rather than the pop-up keypad. 

I'm not sure if a number pad will be added... perhaps as a toggle option in the future. Please keep in mind the current implementation allows you to with one press invert, use the min, default or max values, as well as increment in 1s and 10s (unfortunately for fields that have decimals, that equates to 0.1s and 1s).

closed 2022-10-11 by pfeerick