EdgeTX/edgetx #2520: Model Setup "Throttle Source" not working with CH selections.

electricon, 2022-10-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When powering up a Radiomaster TX16S  transmitter or switching models, the Throttle Position Warning is by default set to the Throttle stick. The Throttle Source can be changed in the Model Setup menu. It works to change the source to S1, S2, LS or RS, as the throttle warning moves to those inputs. If selecting any of the available Throttle Source "CH" selections, the throttle warning stays attached to the Throttle stick position. It doesn't appear to function correctly with "CH" selections. The CH selections become available as Channels are added in the Mixer menu.

### Expected Behavior

I expect when selecting a Throttle Source CH (channel 1-4, etc.), that the Throttle position warning will move to the selected channel output value and that the throttle stick input will no longer be part of the warning equation.

### Steps To Reproduce

1. Goto Model Setup menu
2. Change Throttle Source to (for example) CH2
3. Turn off transmitter
4. Move throttle stick to mid position
5. Turn on transmitter
6. Observe "Throttle Warning- Throttle not idle" on display
7.  Move channel 2 Stick/Output to it's -100% position (Min)
8. Note that Warning doesn't clear
9. Move Throttle stick to -100% (Min)
10. Note that the Warning is now clear

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-10-12:
Ok, that is probably a different problem of sorts - CH# is not actually valid as a throttle source - those are the mixer outputs - which you'll see if you create another mix channel - suddenly you get CH5 ;) So your throttle source options on the TX16S are actually only the throttle stick (which you defined by setting the mode setting at the bottom of the radio setup page),  S1, 6P, S2, LS and RS. That mixer channels probably shouldn't be shown there as I don't think it's valid for _outputs_ to be used as the throttle input source

electricon, 2022-10-12:
The flexibility of EdgeTX is one of the wonderful features of the operating system. It actually may have some usefulness for some people to have Channels available as a source. I personally have never needed to use Channels but wanted to notify the developers  about the issue.

pfeerick, 2022-10-12:
Indeed... but I don't think it is valid for the throttle settings - RF output channels are not input sticks / pots ;) And there is nothing stopping you using an output as the input for another mix in the mixes section anyway.

raphaelcoeffic, 2022-10-12:
Indeed, throttle check only uses the stick, nothing else. If that does not work for you, you need to turn it off.

pfeerick, 2022-10-12:
While that is true, I think the real bug is that invalid options are being shown... whether that can be readily fixed is what I'm wondering though... for 2.9 ;)

raphaelcoeffic, 2022-10-12:
It’s not invalid, it works very good for throttle timer for instance, just not for warnings.

pfeerick, 2022-10-12:
Just so I know we're talking about the same think... you think it's valid that CH1-4+ are shown here? 

https://user-images.githubusercontent.com/5500713/195251688-4b412d9e-de88-4b0d-b6fe-2f5d5caed88d.mp4

raphaelcoeffic, 2022-10-12:
> you think it's valid that CH1-4+ are shown here?

Absolutely! I use this for throttle timer on my motorised gliders, as I have the engine on a slider + switch (LS 50% offset 50% + conditional on `!SA-UP`). But as I said, it does not work for throttle warnings, just timer + `throttle%`.

raphaelcoeffic, 2022-10-12:
this is what the code says about channels selected as throttle source:
```c++
  // throttle channel is either the stick according stick mode (already handled in evalInputs)
  // or P1 to P3;
  // in case an output channel is choosen as throttle source (thrTraceSrc>NUM_POTS+NUM_SLIDERS) we assume the throttle stick is the input
  // no other information available at the moment, and good enough to my option (otherwise too much exceptions...)
  uint8_t thrchn = ((g_model.thrTraceSrc==0) || (g_model.thrTraceSrc>NUM_POTS+NUM_SLIDERS)) ? THR_STICK : g_model.thrTraceSrc+NUM_STICKS-1;
```

pfeerick, 2022-10-12:
> So I believe this is a deliberate choice by the implementor.

Yeah, clearly, but still seems wrong... it should either only give valid options, or actually do what it's told.... And I see what you're getting at now... this allows throttle timer, etc to work with complex throttle/multi-throttle input setups. 

I know if the throttle reverse check is ticked, it (warning) expects the throttle stick up physically up instead of down, not sure about anything else. 

Not sure there is going to be a good solution to this then other than to disable the throttle warning screen since doing it automagically would be a false sense of security. Forcing the user to consciously doing it, knowing it is a known limitation, at least ensures awareness of the decision they made. Maybe another reason to have that custom warnings PR in 2.9, since if I understood the goal there it would allow for custom pre-flight checks.

electricon, 2022-10-12:
A comment was made earlier that throttle check only uses the stick, nothing else. This isn't entirely true. When a slider or pot is selected as the throttle source, the throttle stick position is ignored. It's only when CH# is selected that it only uses the throttle stick position.

pfeerick, 2022-10-12:
As has already been explained, it is only the throttle warning that fails when set to an output channel (CH#). Other uses for the the throttle source, such as TH trigger for timers do work.

So at the present time, the only option will be to disable the throttle warning check, as you are asking for an impossible check atm.

raphaelcoeffic, 2022-10-13:
Let's make an attempt at implementing this in `2.9`.

TODOs:
- run mixer calculations step-by-step to get channel values
- take channels into consideration when checking throttle warning
- extend to more channels (not just `CH1` to `CH4`)?

pfeerick, 2022-10-13:
> extend to more channels (not just CH1 to CH4)?
CH5 will show if you add a CH5 mix, so I _think_ that might already be done ;)

AlphaCharlyX, 2024-03-29:
Why it is not possible to choose a switch as the throttle source? This would solve the whole problem.

As a glider pilot I usually use a 3-position switch for my throttle because the throttle stick is used for flaps or butterfly. Inwould be happy if throttle-warning would work with switches and not only with the sticks and sliders.

J-Sorenson, 2025-02-07:
> Why it is not possible to choose a switch as the throttle source? This would solve the whole problem.
>
> As a glider pilot I usually use a 3-position switch for my throttle because the throttle stick [...]

I think you would just use a regular switch warning.  The "Throttle" warning should not do too much guess-work versus assigning pot/switch warnings, but perhaps the intent was to support beginners with the most important warning.  With that in mind, I think it's reasonable for a user to implement their own warning if they are experienced enough to customize the throttle logic.  I am concerned that the warning treats the  "CH" selections differently without easily communicating this with the user.   Invisible logic is something that should be minimized, but removing the "CH" option may break existing models.  Are people are using the "CH" option to drive the timer?

pfeerick, 2025-05-18:
> Are people are using the "CH" option to drive the timer?

This is exactly what Raphael said he is doing above at @ https://github.com/EdgeTX/edgetx/issues/2520#issuecomment-1275583421

I think the underlying issue is still one of that the throttle check is done before the mixer is running (thus it isn't possible to check the state of the output channel , but I think the code in `isThrottleWarningAlertNeeded()` might be a bit more amenable to this being improved on.