EdgeTX/edgetx #2522: LVGL Flex Layout: finish screen conversions

raphaelcoeffic, 2022-10-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Release `2.8` is still considered a "transition release", as it is already using LVGL, but some screens are still using the old layouts where every control is set at specific coordinates, and most of the screen is completely recreated on each and every change.

### Describe the solution you'd like

The following screens still need to be converted to flex layout:
- [ ] Global functions
- [ ] Curves (on-going work)
- [ ] Global variables
- [ ] Logical Switches
- [ ] Special Functions
- [ ] Custom scripts
- [ ] Telemetry

mha1, 2022-12-16:
Hi, can you please check https://github.com/EdgeTX/edgetx/issues/2818 and point me in the right direction how to fix this? See comments for what I tried and failed miserably.

philmoz, 2023-01-20:
Global / Special functions - #2992 
Logical Switches - #3019 

I was going to look at telemetry next if no one else is doing it.

pfeerick, 2023-01-20:
I'm planning to at looking those two PRs and also the Curves one early next week... have just been giving them time for the discussions to finish ;)

philmoz, 2023-01-21:
Telemetry - #3070

philmoz, 2023-01-21:
Custom mixer scripts - #3071

Eldenroot, 2023-01-21:
Are these all non-converted screens? Any new finding?

gagarinlg, 2023-01-21:
I think that's it

philmoz, 2023-01-21:
There are still some pages using the FormGridLayout class and direct painting into the bitmap buffer:
- Statistics & Debug
- Logical Switches Monitor
- Theme Details
- Spectrum Analyser (footer)
- Select Template Folder

Eldenroot, 2023-02-03:
Good job, only few issues left from the first post!!!

pfeerick, 2023-02-05:
With the merge of the custom script page, this brings to an end the bulk of the LVGL conversion for colorlcd. There is still more work to be done, and improvements that can be made, but this will now mostly flow on from this conversion effort.

closed 2023-02-05 by pfeerick