EdgeTX/edgetx #2530: Selecting MPM Protocol will stuck in certain point

richardclli, 2022-10-12:
Radio: TPro
Firmware: 2.8 latest

Uploading QQ视频_bb776d37fd8261dce9313f39376194b51665587626.mp4…

richardclli, 2022-10-12:
https://user-images.githubusercontent.com/18414861/195384521-b7d870f2-f1e9-4c5e-9023-9e76165361ea.mp4

richardclli, 2022-10-12:
More discovered.  This happens with 2.8 bootloader with 2.8 firmware.  If changed to 2.7 bootloader and 2.8 firmware the problem gone.

richardclli, 2023-02-24:
This is report by others.  Maybe the module just not switched on?  Or failure.

pfeerick, 2023-02-24:
Yeah, it is possible something went wrong with the communications ... I had the MPM go completely stupid the other day when adjusting the fine tune very fast (instead of RX number which is what I intended to change)... it dropped communication and I needed to power cycle the MPM to make it behave.

richardclli, 2023-04-18:
I think this is fixed by #3112

closed 2023-04-18 by richardclli