EdgeTX/edgetx #2540: 2.8.0-RC3 Issue Tracker

pfeerick, 2022-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

stuff breaks

### Expected Behavior

stuff shouldn't break

### Steps To Reproduce

use it

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+

### Anything else?

Long live OpenTX!

Mmcb23, 2022-10-24:
Hi this release seems to have stalled. When can we expect the final 2.8?

jfrickmann, 2022-10-24:
#2617

pfeerick, 2022-10-24:
> Hi this release seems to have stalled. When can we expect the final 2.8?

Once the major regression or bugs are ironed out.

gjhaynes, 2022-10-25:
It's a shame that the word "expect" gets used so much more than "respect" these days. Peter, you and the rest of the devs are doing a phenomenal job! One simply has to look at the milestones to see that the next release is hardly "stalled"! Take as long as you need to get 2.8 stable and ready for prime time. GeoffH.

Mmcb23, 2022-10-25:
I agree the team is doing a brilliant job, and I meant to malice.
All I was looking for is some indication ss to when these last couple of 2.8 items would get sorted, so the release can get out there to the wider community. I for one am eagerly awaiting 2.8 in order to migrate from OpenTx.
Keep up the good work guys.
RegardsMick

Sent from Sky Yahoo Mail on Android

gagarinlg, 2022-10-25:
> Hi this release seems to have stalled. When can we expect the final 2.8?

What do you prefer:
1. quick release with a bunch of known and unknown bugs in it, because of the major overhaul we did
2. wait a little but longer and get something everyone can be comfortable with when flying

We, as in the people contributing to EdgeTX, so this in our spare time, unpaid. Sometime live gets in the way and sometime someone just needs a break. It takes as long as it takes to reach a state where we are willing to finalize it. Maybe tomorrow someone finds a huge problem, then it will be delayed again.
Generally speaking, what is left to be done is not as significant as some bugs we had in the beginning and the frequency of error reports has gone down. So we are still on the way.
If you want to help, so that releases can be done quicker, you have three options:
- fix bugs
- confirm issues reported by others and try to find more details to help fixing the bugs
- install the firmware and test it, to help finding bugs, but make a backup of you SD-Card first, so that you can revert back to the older release without issues.

gagarinlg, 2022-10-25:
In the light of that message from you, which came in while I was writing my message, my message seems to be a bit harsh. Sorry for that.
The general message still stands, though.

Mmcb23, 2022-10-25:
Having been a commercial developer for over 40 years before I retired I fully understand the need for patience while bugs are fixed I totally get that.
I for one would not consider taking a pre-release out to the field and risking one of my models, or the safety of other flyers around me.
My original question was simply to help me to decide whether to go with 2.7.1 and upgrade later or wait for 2.8.
I hope I have not upset anyone by my comments.

Sent from Sky Yahoo Mail on Android

gagarinlg, 2022-10-25:
I meant bench testing 2.8 rc3, not flight testing it. 😉

pfeerick, 2022-10-25:
Oi, I flight test the RCs (no quads have been harmed yet 😇) And remember, it is an RC, not a nightly, so there is some level of confidence in the core functionality working properly... you might just be unlucky enough to make the UI crash or otherwise misbehave (which most people don't fiddle with during flight, so while not _good_, will hopefully not result in any loss of control). But the more the merrier, all the better to find and squash the bugs. 

No offence was taken here. If I were to follow my own advice/rule of thumb for updates for *any* software, I would go to 2.7.1, and then wait for the first point release of 2.8... :grin:

richardclli, 2022-11-02:
I will flight test latest 2.8 branch with EL18 and OMP m2 the coming weekend.  Hopefully the radio is stable and my m2 won't crash.  Well, I have some confident in it anyway.

richardclli, 2022-11-02:
BTW, please add Issue #2666 to the list as well.  I think this is essential, as I spot something unusual before this fix.  I will not rey a real flight without this fix.

Eldenroot, 2022-11-05:
Really huge step forward for last 3 weeks sice RC2 release!

Many bugs fixed; I really like how the community reports new things from testing phase.

I think EdgeTX 2.8 can hit 400 closed issues for this release, we are close :)

[image]

**Thank you EdgeTX team and community, you are wonderful!**

closed 2022-11-06 by pfeerick