EdgeTX/edgetx #2544: Model delete dialog background

banyaszg, 2022-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The background of the delete model confirmation dialog is the same as the button fill color. So the outline of the button is invisible. It is just a minor inconsistency in appearance.

### Expected Behavior

The dialog background should be different to see the buttons.

### Steps To Reproduce

1. It was tested with default theme.
2. On the model selection page select a model to delete.
3. A confirmation dialog pops up.
4. There is no outline for the non-focused button.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-10-13:
@banyaszg I think the issue is that dialogs don't use the regular form background, which make the button have the same color.

banyaszg, 2022-10-13:
I think so. However I didn't make a full check this time, just ran into the problem and recorded it.
(I tried to setup a model.)

raphaelcoeffic, 2022-10-13:
> I think so. However I didn't make a full check this time, just ran into the problem and recorded it. (I tried to setup a model.)

You should see it in basically all the pop-up dialogs that have buttons. So I guess the issue should be solved in a generic way by adding the proper background for all of them at once.

pfeerick, 2022-10-14:
Or change the color of the buttons... (where have I heard that before? :grin:) It is also theme specific - as while this is still not really visible on the actual transmitter, the button is at least a different color to the dialog background.

[image]
[image]

philmoz, 2023-03-02:
I think this is fixed in 2.8.1.

JimB40, 2023-03-03:
Nope.
There are few places in UI where Modal window (pop-up) contains generic data like labels, text or buttons.
In those pop-ups background should be SECONDARY3 not PRIMARY2 

Checked 2.9 nigtly
1. Pop-ups with proper styling

[image]

[image]

2. Pop-ups with wrong styling I have found

[image]

[image]

[image]

[image]

closed 2023-03-22 by pfeerick