EdgeTX/edgetx #2546: Variometer sound

djabmal, 2022-10-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Edgetx 2.8 RC2 variometer sound is not working .
This works fine in 2.7.1

### Expected Behavior

having a proper variometer sound

### Steps To Reproduce

1. in telemetry section Vspeed is on  and the variometer is not beeping 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

edgetx 2.8 RC2

pfeerick, 2022-10-14:
I'm using the latest nightly build, not rc2, but there haven't been any significant changes as far as I'm aware. Please check your SF to enable the variometer is configured correctly still. Even with a zero value, you should be getting the 'neutral tone'

djabmal, 2022-10-14:
i tried it again with rc2 the vario won’t work although it works with 7.1

pfeerick, 2022-10-14:
Ok, probably need a copy of your radio and model folders to see what is going on (zip the folders up, and upload them to the issue - don't email, as github ignores attachments) in case there is something that didn't convert or work in the config file. As I can't reproduce that here atm.

djabmal, 2022-10-14:
ok you are right its working with rc2 but not with rc3 even the latest version .do you need the sd card content of my radio or just models?

pfeerick, 2022-10-14:
Should just need the models/radio folders... sd card content probably won't be relevant in this case. I'm using today's nightly and it works, but will also try with rc3 just to be sure.

djabmal, 2022-10-14:
hello these are the models folder

pfeerick, 2022-10-14:
You need to upload them to the issue - don't email, as github ignores attachments ;) Will need the MODELS _and_ RADIO folder for completeness.

djabmal, 2022-10-14:
sorry i don't know how to upload the files

gagarinlg, 2022-10-14:
Just drag and drop them into the text box

djabmal, 2022-10-14:
nope its not working ( "we don't support this type of files"

gagarinlg, 2022-10-14:
Please zip the files together

djabmal, 2022-10-14:
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/9784345/RADIO.zip)

vbabiy, 2022-10-15:
I am having the same issues, I am on RC3 as well

pfeerick, 2022-10-17:
I finally managed to reproduce this locally, but I'm not sure at the moment that will help you guys.

For some strange reason, if I change from internal ELRS, to external ELRS on the same model, without power cycling the radio, even though telemetry is coming in (again), and no other settings change, I get vario beeps with the internal module, and no vario beeps with the external module. Power cycling does not change this situation.

vbabiy, 2022-10-17:
@pfeerick I can also confirm this work on the internal module but not external.

ruedli, 2022-10-24:
I have tested on RC2.8.0-rc3 self compiled (noheli, with LUA mixer scripts) and can confirm that the vario sound worked for me (as well as display values of ALT and VSPD). I used an internal module (4-in-1) on radio master TX16S with a FrSky S8R, with high-precision vario sensor attached over S.Port.

I configured the vario sound in the respective system settings and configured a switch (SA) to switch the vario "on", "off" and "on with playing out readouts of altitude". (These were already present in the model I tested with, did not change anything).

PR #2653 (2022-10-31): fix(telemetry): sensor processing with legacy polling disabled
## Issue description

External modules using the new module interface (direkt polling of telemetry based on driver) disabled legacy telemetry polling (S.PORT shared pin) to avoid confusing the telemetry parser. However, this fix (see #2504) was implemented poorly and removed sensor processing, which is done in the same function.

**External** modules / protocol impacted:
- `ACCESS`: `R9M ACCESS`
- `CRSF`: Crossfire, Tracer and ELRS

It would impact basically anything that is related to sensor processing:
- Vario sound (#2546)
- RF  [... cut]
commit b61e2d8d (2022-10-31): fix(telemetry): sensor processing with legacy polling disabled (#2653)

Fixes #2546
Fixes #2591
commit bcf145c8 (2022-11-02): fix(telemetry): sensor processing with legacy polling disabled (#2653)

Fixes #2546
Fixes #2591
closed 2022-10-31 by pfeerick