EdgeTX/edgetx #2549: Remove flashing in bootloader mode from windows companion

gagarinlg, 2022-10-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Obviously Windows does brick the radios more often than it does not brick the radios when flashing via the extra drive provided by the bootloader.

### Describe the solution you'd like

This should not be done in the windows build of companion. Just open a pop up to tell the user to use DFU mode or flash from the SD card.

raphaelcoeffic, 2022-10-13:
We would obviously need to re-implement this method differently, as shown by the UF2 method (see https://github.com/EdgeTX/edgetx/issues/910).

In the meantime, disabling that method seems to be the only way.

pfeerick, 2022-10-13:
Funny you say that, I was going to mention that enhancement issue too 😂

elecpower, 2023-03-12:
We are using a very old version of dfu-util for windows installs that has been uploaded to the project repo https://github.com/EdgeTX/edgetx/tree/main/companion/targets/windows rather than sourced from devs.

pfeerick, 2024-07-06:
We removed the firmware drive, so this has effectively been done.

closed 2024-07-06 by pfeerick