EdgeTX/edgetx #2556: Curves names

Billy60, 2022-10-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Could it be possible to name each curve with more than 3 char ? 
Currently it is difficult to use them with a name which has no meaning because of being too short.
Thanks

### Describe the solution you'd like

Let it be possible to use for the name up to 10 characters, min would be 8.

chronowarper, 2022-10-18:
Second this request. Longer curve names was was supported in older OpenTX firmware and it was regressed into just 3 characters around 2.0 or 2.1. 

I can't make anything meaningful out of 3 characters.

Eldenroot, 2023-01-01:
I wanted to create a new ticket but I have found this one.

Would be nice, because sometimes you need to add more characters than just 3. This affects more places - not just curves.

[image]

JimB40, 2023-01-01:
Longer curves names affects not only curves tab and curves  editing screen but few other screens where they are used as parameter.
Doable but I'd postpone it until all LVGL transition is done not to change previously done work.

Eldenroot, 2023-01-01:
I understand, anyway this would be a handy improvement. So v3.0 is the earliest milestone :)

pfeerick, 2023-01-02:
Possibly. In some places, they would be shown longer, and in others, they would just be clipped until the UI is altered to suit that longer length. I think the reality is they need to be made longer and then UI updated to catch up, otherwise, the UI rework may not take the length into account properly.

Eldenroot, 2026-08-29:
This one only for COLOR radios, right? I like this idea, but I think it is related to more parts of UI than just curves.

chronowarper, 2026-08-29:
I assume this was filed for black and white display radios. 

The earlier versions of OpenTX allowed for more than 3 characters in throttle curve but starting somewhere in version 2.xx, the curve names were reduced to 3 characters.

I used to be able to make meaningful names to reflect having different props, battery sizes, acro vs cinematic flight.  Like 4s_gemfn  or 6s_hq_cine

Can't make useful names out of 3 characters.

Eldenroot, 2026-08-29:
yes, 3 symbols are just not enough..