EdgeTX/edgetx #2559: Ghost Menu  cannot be operated.

TheIsotopes, 2022-10-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Unfortunately, the ghost menu can no longer be operated with the scroll wheel since 2.8 RC1, but everything still worked in 2.7.

### Expected Behavior

working ghost menu

### Steps To Reproduce

...

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-10-15:
@tonycake would you mind to please have a look at this? I believe this needs to be converted to the new paradigms used with LVGL & event handling.

pfeerick, 2022-10-15:
The ghost menu wlll show up, so comms with the module are fine, it's just there is no response to input (roller, ENT)
[image]

PR #2618 (2022-11-05): Fix ghost module remote menu for 2.8
This is a quick fix for Ghost module remote menu . Tested only oon TX16S, unsure of behavior on radio without rotary encoder like nv14 (but also unsure how it works today on those)

This fixes #2559
closed 2022-11-05 by pfeerick