EdgeTX/edgetx #2561: Duplication of model in unlabeled category

planemann2000, 2022-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am getting models being duplication in the unlabeled folder in the model select page and not in my favorites, quid copters or the flex categories I have set up in Select Model page. I have deleted duplicated model from this category but duplicates keep returning.  

### Expected Behavior

No duplication of models.

### Steps To Reproduce

Just set up different categories.
Turn radio off
Go to Model select unlabeled category

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

planemann2000, 2022-10-15:
Happens after in update with the latest nightly.

eshifri, 2022-10-15:
I cannot reproduce it.

planemann2000, 2022-10-15:
OK, this is what I did to reproduce the problem. I reinstalled RC3, then updated to the latest nightly 836954ee. Loaded a new model that was in another category, went back to my unlabeled cat. and low and behold I hand duplicate of two of my models. The other 3 cat. are good (no dups.)

planemann2000, 2022-10-15:
Here is my SD card contents, models, firmware and radio
[MODELS (2).zip](https://github.com/EdgeTX/edgetx/files/9792827/MODELS.2.zip)

planemann2000, 2022-10-15:
I cannot duplicate the issue on my main transmitter (Radiomaster mk2) like I can on my test transmitter (Eachine tx16s) but every thing should be identical. I flash my test radio first then my main radio if all is good.

eshifri, 2022-10-15:
Still cannot duplicate it neither on RM TX16S nor in the simulator.

planemann2000, 2022-10-15:
OK that's telling my I may have a bug in my test radio setup, I'm going to format the test SD card and load the main transmitter contents on the SD card and go from there. I will let you know the results.

planemann2000, 2022-10-15:
Ok I thought that I may have found the problem, I discovered the sd card in my test radio was the original sd card that came with the radio, I replaced the card, formatted and copied the content from my main set. After flashing the radio both firmware and the latest nightly I had a duplicate model but in another cat. I flashed the new nightly to my main radio again, which I had done earlier, Now I have a duplicate in unlabeled cat. The dups. appears after a few model changes in different categories, back and forth.

dlktdr, 2022-10-16:
Can you delete labels.yml and boot it up again, see if it comes back that way too?

planemann2000, 2022-10-16:
I deleted labels.yml rebooted, no dup showed up, reflashed 836954ee still no dups. So far so good. but the labels.yml keeps reinstalling itself on boot up. I'll keep a watch on this. I have been deleting the extra models dulicate in Select model page when a dup. shows up, is that the correct procedure?

eshifri, 2022-10-16:
labels.yml is generated on boot if it does not exist, so this is OK.
Deleting model does not delete the file - it just removes it from the labels.yml.
Upon labels.yml regeneration these files were moved into UNUSED folder.

dlktdr, 2022-10-16:
> I deleted labels.yml rebooted, no dup showed up, reflashed [836954e](https://github.com/EdgeTX/edgetx/commit/836954ee4bb2886a25e20248adc867e135566164) still no dups. So far so good. but the labels.yml keeps reinstalling itself on boot up. I'll keep a watch on this. I have been deleting the extra models dulicate in Select model page when a dup. shows up, is that the correct procedure?

Yes is should be automatically regenerated on boot, it will do this by opening every model and reading all the details. It can be very slow if you have many models on your radio which is why it's not done on every startup and the important details are kept in labels.yml.

I did find a potential cause for this, if you change some settings (including labels) in your current model and hit duplicate without waiting the apx 5-10sec (not sure exactly how long, it's a while) before it writes to the file. Then none of those changes would propagate to the duplicated model and thus end up in other categories on reboot. 

If you can test PR #2579 would be appreciated.

planemann2000, 2022-10-16:
Not sure how to test PR#2579, I will leave that up to more knowledgeable people. I haven't tried to duplicate one of my existing model since installing 2.8.0 RC1.

gagarinlg, 2022-10-16:
> Not sure how to test PR#2579, I will leave that up to more knowledgeable people. I haven't tried to duplicate one of my existing model since installing 2.8.0 RC1.

https://github.com/EdgeTX/edgetx/actions/runs/3260337550
at the bottom of the page is a assets download link. Pick the right firmware for your radio from that zip file and install it on your radio. Create a backup of your SD card first.

pfeerick, 2022-10-17:
It looks like Buddy is cooperating with this PR, so that is the easiest way to download the bin/flash the transmitter.
https://buddy.edgetx.org/#/dev/flash/pr?version=pr-2579@241756b4882b1197d4a86341b5b3b891c69e0cd0&target=tx16s

planemann2000, 2022-10-17:
OK thanks I have the firmware flashed on both of my radios, looks good so far. I tried every way to get the dups. to show up. No dups yet.

closed 2022-11-05 by planemann2000