EdgeTX/edgetx #2563: Internal rf “off” setting not save

marinehm, 2022-10-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have turned the internal rf setting to off several times.  Turn off the radio and turn it back on.  Go into radio settings and the internal rf is back on. My quad flys fine.  Internal rf in the quad settings stays off.  Where I’m having issues and trying to pinpoint the origin of the error.  I have a RM TX16S with EdgeTX black pearl version.  Updated ELRS to 3.0  When trying to use EdgeTX companion and connect to the radio wifi.  Wi-Fi’s enabled for a short period then get disconnected and error code: ?:0:attempt to index field’?’(a nil value)

### Expected Behavior

Radio settings (internal rf “off”) to remain off when set to off.  Even after turning the radio off and back on again.

### Steps To Reproduce

Enable wifi using the ELRS LUA.  Connect to HappyModel ES24TX pro micro to pc via ELRS configurator.

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

[image]
[image]
[image]
[image]

pfeerick, 2022-10-15:
That is normal, and expected behaviour. If you want the internal RF to be off, you should be turning it off in the model settings. The 'HARDWARE' page setting is for changing what the default internal RF is, not for turning it off. We should probably remove the 'OFF' option there to stop people thinking they can turn it off there. 

Also, please try to stick to one issue per, well, issue... otherwise multi-issue uses may not get everything resolved when a fix for one part of it closes the issue.

raphaelcoeffic, 2022-10-16:
The other issue is already documented in #2436

closed 2022-10-16 by raphaelcoeffic