EdgeTX/edgetx #2565: total loss telemetry

stubugs, 2022-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

t16s running 2.8 rc 3
crsf 6.19
flying  ar wing with 2.7.1 10 miles no problems on any model 
tried 2.8.rc3 lost all telemetry even signal strength on main top right corner at about 2 miles
 yet in goggles lq 1;100 
but tx reads as if reciever not connected yet still full control and inav lua script stopped recieving data this was not a problem in 2.8 rc1 as you can see by sig quality in 2,8 rc1 lowest tqly 98
2.8 rc 3   0 lowest yet in googles lq 1 100
[image]
[image]

### Expected Behavior

keep telemetry as in 2.7.1

### Steps To Reproduce

flew out as i have before same place 
at  about 2 miles all tx indications ie telem sig strengh stops as if reciever not conected but still flew home 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

in tx quality log from 0 to 100 through out the flight yet lq in goggles as normal around 96 to 97 and controll was still there and no fail safes

EhAye, 2022-10-17:
is it a matek h743-wing by any chance?

stubugs, 2022-10-17:
No it's a matek 405 wing works fine on 2.7.1 10 miles plus

closed 2023-03-31 by stubugs