EdgeTX/edgetx #2566: EMERGENCY MODE error when exiting mode (themes) touch color LCD. RМ TX16S 2.8.0-RC3

Alex6892, 2022-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EMERGENCY MODE error when exiting mode (themes), touch color LCD.

### Expected Behavior

Exit

### Steps To Reproduce

RМ TX16S 2.8.0-RC3
Attached video.

https://user-images.githubusercontent.com/115475444/196026364-ead435cc-c570-4766-b2ea-3092e42f92c7.mp4

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

RМ TX16S 2.8.0-RC3

raphaelcoeffic, 2022-10-16:
Fix has been pushed to `main` & `2.8`

closed 2022-10-16 by raphaelcoeffic

Alex6892, 2022-10-16:
Sorry for my English.
I'm a little dumb.

gagarinlg, 2022-10-16:
Most people here are not english native speakers. Nothing to be ashamed of

pfeerick, 2022-10-16:
Definitely nothing to be ashamed off (English proficiency) ... and the bug report was perfect with a video to easily reproduce ;)

commit adab4dcd (2022-10-16): fix(color): check `theme` before accessing it

Fixes #2566
commit 27616254 (2022-10-16): fix(color): check `theme` before accessing it

Fixes #2566