EdgeTX/edgetx #2568: EdgeTX Companion 2.7.1 cannot find SD card on Mac OS.

curuvar, 2022-10-16:


raphaelcoeffic, 2022-10-16:
Fixed in #2564

closed 2022-10-16 by raphaelcoeffic

magimat, 2022-11-03:
I still have this exact problem, even with RC3 of the companion app which renders it useless at this point :(

Does this mean I should try to build the companion app directly form latest sources for now?

How far away is RC4 ?

thanks

pfeerick, 2022-11-03:
Next few days. 

Companion builds are generated automatically when there are any changes to Companion - you can see the latest builds in the Actions tab at the top of the page. Try https://github.com/EdgeTX/edgetx/actions/runs/3376525087 for the latest 2.8 branch build (scroll to the bottom for the build artefact/download).

magimat, 2022-11-03:
awesome, it works now. I will be using this build until RC4

thanks