EdgeTX/edgetx #2571: elrs setting up working extremely slow while yaapu script in on running.

xixijoe, 2022-10-16:
I found elrs lua script for setting is extremely slow while yaapu is running on the edgeTX system. and elrs setting back to normal if yaapu script is turned off.

raphaelcoeffic, 2022-10-16:
@xixijoe @Flierman1945 which version are you guys using? Have you tested with `2.80-rc3`?

Flierman1945, 2022-10-16:
> I found elrs lua script for setting is extremely slow while yaapu is running on the edgeTX system. and elrs setting back to normal if yaapu script is turned off.

Hello,
I did find if Yappu was running, elrs script would not even start. Closing yaapu and the script started and remained working if yaapu was restarted. But in the ens, I took yaapu off.

pfeerick, 2022-10-17:
@Flierman1945 Please retest with 2.8.0-rc3 if you get a chance, as it should be resolved now.

xixijoe, 2022-10-17:
> @xixijoe @Flierman1945 which version are you guys using? Have you tested with `2.8.0-rc3`?

it was 2.7. I have not tried 2.8

Flierman1945, 2022-10-17:
I am on 2.7.1. Going flying tomorrow and will do the update to 2.8rc3 after and report.

pfeerick, 2023-02-19:
Closing as I'm pretty sure this was resolved.

closed 2023-02-19 by pfeerick