EdgeTX/edgetx #2576: Some custom model sounds not working

shkup, 2022-10-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When using the model folder with sounds in it, they are only partially work. SD switch does work and play tracks properly (for all states: up, mid down) but SF does not. (The wav files were copied from the SOUNDS\en folder so they are perfectly valid).

### Expected Behavior

When SF is up or down and there are wav files present in `SOUNDS\en\MODEL_NAME\SF-up.wav`, `SOUNDS\en\MODEL_NAME\SF-down.wav` they should be played according to the SF switch state. 

### Steps To Reproduce

1. Locate `SF-up.wav` and `SF-down.wav` files under `SOUNDS\en\MODEL_NAME`
2. Select the model
3. Switch SF up and down

### Version

2.7.1

### Transmitter

FrSky Q X7 / Q X7S (ACCESS)

raphaelcoeffic, 2022-10-16:
> SF switch state

@shkup can you please check the configuration of `SF` in the radio hardware config?

eshifri, 2022-10-16:
@shkup Can you please clarify the "model folder with sounds in it" ?
There is a SOUNDS folder for sounds. WAV files in the MODELS folder are not used.

shkup, 2022-10-16:
@eshifri I work according to `https://doc.open-tx.org/manual-for-opentx-2-2/advanced-features/audio#model-folder-sound-files` switches specification. I know it's opentx documentation but as far as I know the same should work in EdgeTx as well.
@raphaelcoeffic At the system hardware page, SF is defined under switches as a 2POS switch (as it really is).

shkup, 2022-10-17:
@raphaelcoeffic This feature saves the need of defining special functions (of audio track playing) for each model and every switch state. You open a folder under sounds with the name of the model and put there `wav` files with the <switch name>-<state>naming convention (SD-up, SD-mid, SD-down for instance). 
It works perfectly fine for all switches except the `SF` switch. This is legit feature. It seems that there is a bug with this switch.

raphaelcoeffic, 2022-10-17:
Oh! I see! There is no special function! Now I think I understand. I should have better read the OpenTx doc link you posted 🙄

pfeerick, 2022-10-17:
Yeah, there's a few hidden gems like that still. I'll have a look at that in the morning unless you want it... as I was poking around there a little while ago so have at least some idea of what's going on there (I think!).

pfeerick, 2022-10-18:
@shkup Can you check what happens when you call the files SE instead of SF? It looks like we have inherited a bug in OTX where the folder based naming doesn't taking into account a skipped switch. i.e. Using OTX Companion 2.3.15, when I flick switch SF it instead wants to play `SE-up.wav`, `SE-down.wav`

shkup, 2022-10-18:
@pfeerick Changing to `SE` switch works but it flips between up and down state. Of course, you can change the files accordingly. It's definitely a bug IMO. For the meantime I'll use the workaround.

pfeerick, 2022-10-18:
It's definitely a bug, but since I don't have the hardware just wanted to confirm the hardware behaves the same. 

@3djc Are you able to have a look at this, as you mentioned you do have an X7 still? :grin: I'm pretty sure both OTX and ETX share this bug, as OTX Simulator was showing this behaviour. It seems the fact the X7 doesn't have a SW E wasn't taken into account somewhere with the model audio indexing, so the SE audio is being played when SF switch is moved. 

Any idea what the rationale was for skipping E and G? Was this to make it so the switches that were present were the same as the X9 series?

shkup, 2022-10-18:
@pfeerick Please notice the switch states are flipped as well (Up is Down and vice versa)

raphaelcoeffic, 2022-10-18:
> Please notice the switch states are flipped as well (Up is Down and vice versa)

That is because it is interpreted as another switch, should be fixed at the same time, no worry.

PR #2599 (2022-10-19): fix(x7): use switch indirection to fetch the proper name
Fixes #2576
commit d7598c8e (2022-10-19): fix: use switch indirection to fetch the proper name for switch audio (#2599)

Fixes #2576
commit 7cf2acb3 (2022-10-31): fix: use switch indirection to fetch the proper name for switch audio (#2599)

Fixes #2576
closed 2022-10-19 by pfeerick