EdgeTX/edgetx #2581: Display VFR in range check mode

MikeBland, 2022-10-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

 No

### Describe the solution you'd like

If VFR telemetry value is available, display it when in range check mode.

MikeBland, 2022-10-16:
Yes, that sensor ID. I think it may be on most, if not all, ACCESS receivers. It is in the UNI firmware I've written for several ACCST receivers.
It provides a value from 0 to 100 that is the number of valid frames received during the last 100 frame times. It is actually transmitted as the number of frames lost in that time.

raphaelcoeffic, 2022-10-16:
@MikeBland nice! thx for the info! Would be cool to have that UNI firmware so I can test, indeed. I'm not really into ACCESS receivers...

raphaelcoeffic, 2022-10-17:
I'd like to check the timings of the link quality packages, as I think one would prefer to use it in general as the built-in indicator in case available.

StefanKellerAC, 2022-10-17:
More generally, a telemetry widget could be shown in range check mode. 
So also those self build quality sensors can be shown, graphs can be displayed, ...
Maybe also showing the spectrum analyzer could be handy.

pfeerick, 2022-10-17:
How about the reverse, standalone Lua being able to trigger range check (rather than add complexity to the range check screen)? Then the Lua script can basically do whatever it wants ;)

StefanKellerAC, 2022-10-17:
Yes, but this may be dangerous.
You could trigger range mode accidentically during flight...

pfeerick, 2022-10-17:
True... and you could do the same right now :-P 

At least with a standalone Lua, it would be up to the Lua author to add a safety checks - i.e. confirmation prompts, etc. Plus you would have to explicitly go into the tool as well. 

overlay "range check"? Standalone lua have basically full control over the screen, so can show whatever they want.

raphaelcoeffic, 2022-10-17:
> You could trigger range mode accidentally during flight...

that depends 100% on how you design that script ;-) And by the way, when doing range check, you shouldn't be flying.

StefanKellerAC, 2022-10-17:
ok, I was thinking of an easy to program telemetry widget with a range check button.
A standalone tool would of course be safe.

offer-shmuely, 2022-10-18:
Is there a lia api to move the radio  to range teat?

What will a script do if there also external module on the tx, how it will know which of them to range test?

pfeerick, 2022-10-19:
Not that I am aware of, that is why I was floating the idea that it could be added ;)

MikeBland, 2022-10-19:
> More generally, a telemetry widget could be shown in range check mode. So also those self build quality sensors can be shown, graphs can be displayed, ... Maybe also showing the spectrum analyzer could be handy.

VFR is, I understand, sent by FrSky ACCESS receivers. It is also sent by FrSky ACCST receivers running the "UNI" firmware.
My suggestion is to simply display VFR if available, along with RSSI when in range check mode. No need for users to actually know if their Rx has VFR or then need to set up anything to see it.

There are times when VFR shows quite low even though RSSI looks good as RSSI is often the RSSI of those frames actually received.

Other options, such as displaying a widget, should, I believe, be in addition to a simple display of VFR with RSSI.

offer-shmuely, 2022-10-19:
The way we use range-test on planes, is going far away until the surfaces are not moving correctly.
for this, we need to have an assistant that stand near the plane and wave us when they are not moving smoothly.

when thinking about it, this is exactly the VFR/link-quality indication, so it may be even more usable that RSSI on the range-test situation, since we will not need another person assistant any more.

it need to decide on a generic name, since some call it VFR, some RQLY, and I think that I saw also LQ
but they are all the same concept

pfeerick, 2022-10-20:
The names used is decided by the vendors, not us. So Crossfire and ELRS provided LQ as one of the metrics, Frsky provide RSSI and now VFR, Flysky and Spektrum probably only have RSSI also. 

From my understanding, LQ and VFR of themselves are not of much use for a range test (or model locating) - they are only telling you *how* many of the packets that are sent actually make it to the destination. It's not telling you _what_ the signal strength is, that's what RSSI tells you. But what RSSI isn't telling you, for example, is that while though you might have excellent signal strength, there is some intermittent noise/multipath interference/etc that is making it so 50% of the packets aren't making it from the transmitter to the receiver safely. So by displaying both values together, you can see both the signal strength AND the signal health. Because you could have low LQ/VFR, but good RSSI (since RSSI is based on the packets that actually make it), and you could also have good LQ, but bad RSSI (thus the link/LQ could drop suddenly).

Please correct me if I'm wrong! :)

offer-shmuely, 2022-10-20:
LQ (RLQY/VFR/LQ) is not useful for locator, since when you are getting closer to the lost model, the LQ will show 100% and it will not help to continue with the correct direction.

on the other hand, for range test, this is exactly what you need, since you are going away from model, and like to see how far you can go, and still have surface control, RSSI does not tell you when you loose control, but LQ does,
especially on expressLrs, that they have strange way to show rssi, will -108db say you lost control? will -112db say you lost? will -115db?
but if you see only 20-30% of frames arrive, it can be considered lost.

small correction on the VFR:
Frsky does provide RSSI, on their ACCESS version, not on the ACCST.
but the ACCST have the VFR on the (great!) UNI firmware the Mike did.

pfeerick, 2022-10-20:
> on expressLrs, that they have strange way to show rssi, will -108db say you lost control? will -112db say you lost? will -115db?
but if you see only 20-30% of frames arrive, it can be considered lost.

It's not that it's strange, though, is it? Because of the loRa chipset, it's the packet rate that the user sets that determines that the minimum viable signal is for ELRS, and this is why LQ becomes the single important metric for range. Because like with Ghost or CRSF, the user can choose whether to prioritise range or fresh rate. 

And yes, I should have been more clear that it was ACCESS that VFR is available. and with UNI firmware for ACCST. I thought that would be clearly understood since Mike had outlined that. Which I have started using myself ;)