EdgeTX/edgetx #2591: Telemetry call-outs vanish when upgrading 2.8 RC2 to 2.8 RC3

howels, 2022-10-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

No RF signal critical warnings when model is connected, no telemetry lost warnings when disconnecting battery from model.

On 2.8.0 RC2 this worked as expected, upgraded to 2.8.0 RC3 and the telemetry sound warnings have vanished.

### Expected Behavior

When telemetry is initialised there is usually a "RF signal critical" warning while link stabilises, then when battery is disconnected from model we should hear "telemetry lost".  This happens on RC2 but neither sounds work on RC3.

### Steps To Reproduce

Power on radio.
Power on model - no "RF signal critical" sound
Power off model - no "telemetry lost" sound

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-10-17:
Oh, oh, oh... can you check the telemetry screen... warning values for low and critical alarm, and is the disable alarms checkbox ticked or unticked? If it's ticked that would explain why no prompts. I just noticed on my LR3 with rc3 it was zeros, but unticked, so I get the prompts.

howels, 2022-10-17:
You are correct - RC3 is simply disabling the telemetry alarms - when the alarms are re-enabled it works again.

So RC3 is changing this setting for some reason, this should not be changed on upgrade.

pfeerick, 2022-10-17:
This probably means https://github.com/EdgeTX/edgetx/pull/2251 is doing something it shouldn't be... as that was not intentional. You should have simply gotten the ability to be able to set telemetry values from 0 to 100, in line with ELRS link quality percentage, rather than be stuck with the Frsky RSSI range.

SteveCEvans, 2022-10-17:
Having checked the **Disable telemetry alarms** setting, RC2 reports telemetry loss correctly, but RC3 does not.

This is using a TX16S with ELRS 3.0.0.

raphaelcoeffic, 2022-10-17:
> Having checked the **Disable telemetry alarms** setting, RC2 reports telemetry loss correctly, but RC3 does not.
> 
> This is using a TX16S with ELRS 3.0.0.

Could you please upload the model's `.yml` file?

howels, 2022-10-17:
On the BetaFPV LiteRadio3 Pro this was simply disabling the "disable alarms" checkbox, but on my TX16S this is not happening - the "disable alarms" checkbox is not checked but the alarms are silent. Toggling the checkbox on the TX16S changes nothing.
BetaFPV literadio3 pro is using Internal eLRS whereas the TX16S is using external. Otherwise the eLRS version is the same on both and settings are identical between the two models as far as I can see.

pfeerick, 2022-10-19:
I unfortunately, don't think I have a backup on the LR3 from pre-rc3 for this model (as I swear it flipped the disable option on it's own, which is what prompted me to say to check it). I tried with another model on the radio, but it wouldn't do it. However, with rc1 it shows normal frsky low and critical values, and on rc3 it shows zeros. Until I toggle a value, rc3 doesn't write anything different (compared to rc1) to the file model file, and even then it was just the value toggled (timer minutes elapsed checkbox) and rc3 also won't write `rfAlarms` until I change them off the zero values they were changed to. 

[model02.zip](https://github.com/EdgeTX/edgetx/files/9818242/model02.zip)

howels, 2022-10-19:
Weird thing is that on TX16S I cannot enable telemetry alarms, toggling the checkbox does nothing!

pfeerick, 2022-10-19:
What happened with the min/max values for you there (they are zero for me on TX16 on opening one model, but normal when opening another)... changing that doesn't effect the sound, and the `disableTelemetryWarning` value is correctly updating for that model.

howels, 2022-10-19:
This model has the following settings:
```
disableTelemetryWarning: 0
rssiSource: none
rfAlarms: 
   warning: 75
   critical: 42
```

Should `rssiSource` be set to something else? I have not changed this value.

howels, 2022-10-19:
> What happened with the min/max values for you there (they are zero for me on TX16 on opening one model, but normal when opening another)... changing that doesn't effect the sound, and the `disableTelemetryWarning` value is correctly updating for that model.

All my models show min/max of zero except my eLRS model which has 75/42 set.

PR #2643 (2022-10-31): fix(yaml): RF link alarms
Set warning/critical to default values before reading YAML model file. Otherwise, these values will be 0 if either the old or the new tags are not present, which is the case when coming from 2.7.

Fixes #2591
PR #2653 (2022-10-31): fix(telemetry): sensor processing with legacy polling disabled
## Issue description

External modules using the new module interface (direkt polling of telemetry based on driver) disabled legacy telemetry polling (S.PORT shared pin) to avoid confusing the telemetry parser. However, this fix (see #2504) was implemented poorly and removed sensor processing, which is done in the same function.

**External** modules / protocol impacted:
- `ACCESS`: `R9M ACCESS`
- `CRSF`: Crossfire, Tracer and ELRS

It would impact basically anything that is related to sensor processing:
- Vario sound (#2546)
- RF  [... cut]
commit b61e2d8d (2022-10-31): fix(telemetry): sensor processing with legacy polling disabled (#2653)

Fixes #2546
Fixes #2591
commit bcf145c8 (2022-11-02): fix(telemetry): sensor processing with legacy polling disabled (#2653)

Fixes #2546
Fixes #2591
closed 2022-10-31 by pfeerick