EdgeTX/edgetx #2595: EdgeTX 2.8 RC3 BUG

beastyboyuk, 2022-10-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have the Radiomaster TX16S MKII MAX and 2.8 rc3 installed the radio goes into emergency mode when in the menu only when going throw the settings  thanks John

### Expected Behavior

Edit the model setup without going into emergency mode 

### Steps To Reproduce

When I go into the setup menu and scroll throw the top bar it just good into emergency mode 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

raphaelcoeffic, 2022-10-18:
@beastyboyuk we will need a whole lot more details than this.

Some questions:
- which "setup menu"? `Model settings` or `Radio settings`?
- what does "going throw the settings" mean?

Then, we might need to have your radio/model files, so please ZIP the following directories from your SD card:
- `/RADIO`
- `/MODELS`

beastyboyuk, 2022-10-18:
> @beastyboyuk we will need a whole lot more details than this.
>
> Some questions:
>
> * which "setup menu"? `Model settings` or `Radio settings`?
> * what does "going throw the settings" mean?
>
> Then, we [...]

The model settings sorry happens a lot in curves
[edgetx 2.8rc3 files.zip](https://github.com/EdgeTX/edgetx/files/9811874/edgetx.2.8rc3.files.zip)

beastyboyuk, 2022-10-18:
> @beastyboyuk we will need a whole lot more details than this.
>
> Some questions:
>
> * which "setup menu"? `Model settings` or `Radio settings`?
> * what does "going throw the settings" mean?
>
> Then, we [...]

> @beastyboyuk we will need a whole lot more details than this.
>
> Some questions:
>
> * which "setup menu"? `Model settings` or `Radio settings`?
> * what does "going throw the settings" mean?
>
> Then, we [...]

[edgetx 2.8rc3 files.zip](https://github.com/EdgeTX/edgetx/files/9815499/edgetx.2.8rc3.files.zip)

raphaelcoeffic, 2022-10-21:
@beastyboyuk I can reproduce with your models either. Could you please produce a precise list of steps to make it crash?

beastyboyuk, 2022-10-22:
> @beastyboyuk I can reproduce with your models either. Could you please produce a precise list of steps to make it crash?

Yes got it to do it on video thanks for your time and effort on this issue hang on I can't post video I put it on YouTube and post a link thanks

https://youtu.be/f692QmBlmSw

pfeerick, 2022-10-24:
Ok, so Model Settings -> Curves  -> Edit. Press `PAGE>` or `PAGE<`. Nothing happens, Press `RTN` ... wait for the EM. 

Btw, `TELE` is also working in places it *shouldn't* be also ... i.e. model settings, mixes screen, editing ouputs, etc.

Richrude19, 2022-10-24:
https://user-images.githubusercontent.com/111724494/197441734-7332ffbc-a281-4503-af13-5d6f37f87654.MOV

related to this?

pfeerick, 2022-10-24:
Oh, interesting... so just to confirm the repo case... did that start happening when you scrolled down to that toggle, then pressed `TELE`, and then `RTN`? As that is what is making that happen for me... 🤡

raphaelcoeffic, 2022-10-24:
@pfeerick nice catch! Does it happen in simu as well?

pfeerick, 2022-10-24:
Yup. And it's not unique to that screen... any 'ol toggle, now that the magic of `TELE` has been unleashed... toggle mayhem is jsut one right arrow press away 😆

Richrude19, 2022-10-24:
Starts when the toggle is selected, then continues even after changing state.
Also while in set up (model) if ii edit something then try to change page with the keys exits setup, if I change page on screen it changes.

beastyboyuk, 2022-10-31:
I have tried the latest nightly build and the issue is sill there do we no when its going to be fixed lol. I am so scared at field if I press wrong back button it will go into emergency mode lol.

pfeerick, 2022-10-31:
Soon... the PR with the fix that is mentioned above hasn't been merged yet. If you desperately need that right now, use a build from that PR as it shouldn't EM anymore, but there is still a little more work needed.

raphaelcoeffic, 2022-10-31:
> there is still a little more work needed

I think I managed to demonstrate the opposite 😝

closed 2022-10-31 by pfeerick

pfeerick, 2022-10-31:
Indeed you did... so this will be in the next nightly...

beastyboyuk, 2022-10-31:
> Soon... the PR with the fix that is mentioned above hasn't been merged yet. If you desperately need that right now, use a build from that PR as it shouldn't EM anymore, but there is still a little more work needed.

Am I been stupid ? were do I find this build lol I would love to use it for now

pfeerick, 2022-10-31:
The fix that was mentioned - https://github.com/EdgeTX/edgetx/pull/2642 - can be downloaded when you go to the "Checks" tab for that pull request (and indeed, *any* PR), then click on "Run tests and package firmware", and then scroll down to the bottom to get to the "edgetx-firmware-merge" Artifact (a zip file with all the firmware builds). You can then copy the correct firmware to your radio and flash it. 

At this point though, I would instead recommend grabbing the artifact (zip file) from [this build instead](https://github.com/EdgeTX/edgetx/actions/runs/3360796120), which will pretty much be the next nightly, which includes another fix for another issue related to the hardware keys (although it isn't anywhere near as bad) as this was ;)

beastyboyuk, 2022-11-01:
> The fix that was mentioned - #2642 - can be downloaded when you go to the "Checks" tab for that pull request (and indeed, _any_ PR), then click on "Run tests and package firmware", and then scroll down to the bottom to get to the "edgetx-firmware-merge" Artifact (a zip file with all the firmware builds). You can then copy the correct firmware to your radio and flash it.
> 
> At this point though, I would instead recommend grabbing the artifact (zip file) from [this build instead](https://github.com/EdgeTX/edgetx/actions/runs/3360796120), which will pretty much be the next nightly, which includes another fix for another issue related to the hardware keys (although it isn't anywhere near as bad as this was - more a annoyance)

Thanks very much worked like a charm

PR #2642 (2022-10-31): fix(color): prevent event bubbling in Page and TabsGroup
... otherwise unexpected things happen.

Fixes #2595