EdgeTX/edgetx #2598: Lua widgets: zone width/height not updated after screen options are changed

RC-SOAR, 2022-10-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Running 2.8.0 RC-3, TX16S

I have a simple Lua widget which draws a rectangle according to current zone rectangle. On initial load with default screen options, it reports the zone correctly. However, after changing the screen options, the zone width and height are not updated.

### Expected Behavior

Should update Zone width and height, as in OpenTX.

### Steps To Reproduce

1. Set initial screen options: full screen, with all options ticked. 
2. Run script. It reports zone as expected.
3. Now uncheck all the screen options to release space.
4. Script reports correct x,y, but does not update width and height. (last screenshot)
[image]
[image]
[image]
[image]

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Script code:

```
local defaultOptions = {}

local function create(zone, options)
	return {zone=zone, options=options}
end

local function update(wgt, newOptions)
end

local function background(wgt)
end

local function refresh(wgt, event, touchstate)

  local st = string.format ("w=%s, h=%s, event=%s", wgt.zone.w, wgt.zone.h, event)
  lcd.drawText (wgt.zone.x, wgt.zone.y, st, SMLSIZE)
  lcd.drawRectangle (wgt.zone.x, wgt.zone.y, wgt.zone.w, wgt.zone.h)
end

return { name="paneinfo", options=defaultOptions, create=create, update=update, refresh=refresh, background=background }
```

offer-shmuely, 2022-10-18:
I also have the same issue, on a new widget that i am building, and it have to be on specific resolution

jfrickmann, 2022-10-24:
The widget gets zone data on initialization as a Lua variable. Once it has been passed to Lua, the system can no longer change it. Therefore, you have to re-initialize the model, e.g. by changing models or cycling power. It is a basic design flaw, and not particularly smart IMHO.

offer-shmuely, 2022-10-24:
is it possible to send the info in the "update" api, and let the widget to decide it it like to rearrange the display on screen based on the new value?

lshems, 2022-11-15:
Tables are passed by reference in LUA. This means that the zone tavle, if it is referenced as part of the widget table to be able to use it in the run function, should be updated PERIOD. That would be the correct implementation of LUA principles.

So zone gets passed as table {x, y, w, h} in the create function, and the widget array that is to be returned should reference the zone table.

So the table zone exist somewhere in LUA space, and if the widget table that is returned does only refer to this table, then the table individual elements should be update in the update function if the widget layout is changed so that the sizes change.

local zone = {x=0, y=0, w=100, h=50}
local widget = {referencedZone = zone, otherStuff = "otherStuff"}

print(widget.referencedZone.x)
-> 0
zone.x=10
print(widget.referencedZone.x)
-> 10

philmoz, 2023-01-18:
It would be simple to fix this in the refresh logic by updating the stored zone table before calling the widget 'refresh' function.

It would be more efficient, and probably more logical, to do it in the update code; but this would require calling update on every loaded widget when the screen layout options are changed.

Is there an easy way to iterate over all the loaded widgets?

jfrickmann, 2023-01-18:
Where is the zone table stored?

philmoz, 2023-01-18:
In the Lua registry (https://www.lua.org/pil/27.3.1.html).
It's a sub-table of the table returned by the widget's create function.

jfrickmann, 2023-01-18:
The registry is used by C to store references. E.g. the functions that are called by the system are stored in the registry. But as it is now, the zone table is just returned to the `create` function and not stored in the registry.
It is possible that we could store the zone table in the registry before returning to `create` so we can modify it later - worth a try!
By the way, the registry keys are created with `luaL_ref` instead of the method described in PIL 27.3.1, e.g.:
https://github.com/EdgeTX/edgetx/blob/acd44516d01d82abf3db48d512b34ab4d958ab83/radio/src/lua/widgets.cpp#L237-L240
Using it consistently guarantees uniqueness of the keys returned (https://www.lua.org/manual/5.2/manual.html#luaL_ref).

philmoz, 2023-01-18:
The registry usage in luaLoadWidgetCallback is for when the widget is first loaded and handles the table returned by the script on loading.

When the 'create' method is called the table it returns is then stored in the registry (lua_widget_factory.cpp line 103). Assuming this follows the pattern described in the documentation this will be a table with the zone and options tables as sub-tables.

When the update, refresh or background methods are called this widget table is retrieved and passed into the Lua script as the first parameter.

If you want to try it just add the code below after line 336 (lua_rawgeti call) in lua_widget.cpp. This will update the zone date in the widget parameter passed to the refresh method in the script.
`
  // Update widget zone with current size and position
  lua_newtable(lsWidgets);
  l_pushtableint("x", 0);
  l_pushtableint("y", 0);
  l_pushtableint("w", rect.w);
  l_pushtableint("h", rect.h);
  l_pushtableint("xabs", rect.x);
  l_pushtableint("yabs", rect.y);
  lua_setfield(lsWidgets, -2, "zone");
`

jfrickmann, 2023-01-18:
That code adds the `zone` table to the `widget` table. While it is common practice to do so, it is not mandated. In the current API, `zone` is passed as an argument to `create`, and then it is up to the script writer if she will store it in the `widget` table for later use, before returning `widget` to the system.
You correctly pointed out that ETX uses the registry to store the functions, and the `options` and `widget` tables, so the functions can be called with those table arguments later.
What I proposed, is to also add `zone` to the registry before calling `create`, so ETX can change its values later. Since tables are referenced, it will also change for the user's script, no matter how she stores a reference to the `zone` table.

philmoz, 2023-01-18:
@jfrickmann Thanks, I understand what you are proposing now.

philmoz, 2023-01-18:
I have implemented the solution proposed by @jfrickmann and it works fine - if I print the zone values in a test widget I can see them update when the layout changes.

With this solution the script is not notified of the change in layout - the 'zone' table values just change behind the scenes.

When options are changed the script is notified by the call to the 'update' function.

How would widget scripts know that the layout has changed without constantly checking for changes to the 'zone' table?
Should the 'update' function also be called if there is a layout change?

pfeerick, 2023-01-19:
> Should the 'update' function also be called if there is a layout change?

My 2c is that sounds right - while `update()` is described as being called when user settings are changed, I don't see why it shouldn't be called when any settings pertaining to the widget change... including its viewport size. However, I leave it to Jesper and LShems to have the final say as they're the Lua guys ;)

jfrickmann, 2023-01-19:
Be sure to check that it can handle if a user deletes a field in the `zone` table e.g.
```
zone.h = nil
```

offer-shmuely, 2023-01-19:
The widget have to know about the change, since to reduce cpu load, some of the automated placement calculations done once (in create & updates) 
So, from the widget code, getting an “update()” call will be great

philmoz, 2023-01-19:
I've added this in PR 3060 if you want to test it.

closed 2023-02-20 by pfeerick

offer-shmuely, 2023-02-20:
tested with a widget that had this problem.
and it work great!
thanks

offer-shmuely, 2023-02-20:
I see that it even solve the sizes of the app-mode
so in the future we will not need to do special code for that.
 💯

PR #3060 (2023-02-20): fix(lua) - Store and update the 'zone' rectangle for Lua widgets.
Fixes #2598

Summary of changes:

Implements solution from @jfrickmann in issue 2598:
- Store the widget zone rectangle data in the Lua registry.
- Updates the zone data when the layout changes.

Also calls the widget 'update' function if the zone has been altered.