EdgeTX/edgetx #2600: Home screen in 2.8: revert to stacking LS and EX1 sliders in same column?

RC-SOAR, 2022-10-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In 2.8.0, an extra column is used up for the display of EX1/2 and this reduces the available drawing width. This can in turn cause problems with widgets designed for the wider screen area. 

Is there any chance of reverting to the behaviour in 2.7, where EX1 is stacked on top of the existing LS display? That way, users who upgrade their hardware will not run the risk of breaking their widgets.

[image]

[image]

### Describe the solution you'd like

See above

raphaelcoeffic, 2022-10-18:
Should be possible, just takes someone spending on thoughts on how to do it best.

Eldenroot, 2022-10-19:
I like ot how it was in 2.7

Or better way - you can (from my Spektrum era) disable this trim elementa and hide them. For example if you fly drones, you dont need trims in major cases. This also disablea trim buttons.

raphaelcoeffic, 2022-10-19:
You can hide them already, but they won’t be disabled. They can be disabled one by one in each input though.

Eldenroot, 2022-10-22:
Ok, but I still more like the 2.7 look.  Saving space :)

RC-SOAR, 2022-10-22:
Yes, the issue is widgets breaking when using the non-stacked 2.8 format, due to the reduced drawing width.

gagarinlg, 2022-10-22:
Maybe I will take a look this weekend

closed 2022-10-24 by pfeerick

RC-SOAR, 2022-10-24:
Many thanks!

PR #2623 (2022-10-24): fix(gui): stack vertical slider display instead of drawing them next …
…to each other

fixes #2600
commit 50c81670 (2022-10-24): fix(gui): stack vertical slider display instead of drawing them next to each other (#2623)

fixes #2600
commit a7316603 (2022-10-31): fix(gui): stack vertical slider display instead of drawing them next to each other (#2623)

fixes #2600