EdgeTX/edgetx #2617: Touch is unusable with Lua widgets in 2.8 rc3

jfrickmann, 2022-10-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The touch interface is so slow and unresponsive that Lua widgets with touch are no longer usable.

### Expected Behavior

It should work

### Steps To Reproduce

1. Create a new model from the SoarETX F3K template
2. Go to the second screen and open the Outputs widget in fullscreen mode
3. Try using the touch interface!

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Version 2.8 rc3

raphaelcoeffic, 2022-10-21:
@jfrickmann I'm afraid this is actually the slow UI when using full-refresh. We're taking quite a huge impact because of the screen which is rotated by 180°. So the rotation is in software now. On targets without that software rotation (screen is either straight, or has a display controller that does it in hardware).

jfrickmann, 2022-10-21:
That also explains why the screen is polarized so it blacks out with polarized sunglasses 😕
I guess I will stay on 2.7 then.

jfrickmann, 2022-10-21:
Are there any color radios out there with the necessary hw support?

raphaelcoeffic, 2022-10-21:
Is that because of some drag-drop? Or what is actually so slow? Could you try to check how many frame per second you're getting?

raphaelcoeffic, 2022-10-21:
By the way, the NV14/EL18 has a proper display controller, so it's really faster in terms of frames per second.

jfrickmann, 2022-10-21:
No, it is just what I described above for how to reproduce the problem. 
NV14 is a very different radio with a completely different screen size. Is there anything with a c480x272 screen that works?

raphaelcoeffic, 2022-10-21:
Jumper T18 has the screen in correct direction.

jfrickmann, 2022-10-21:
How do I measure the frame count?

raphaelcoeffic, 2022-10-21:
You could print something on debug after each LUA frame. Don't forget to compile with `-DDEBUG=YES` and set the port in radio/hardware.

jfrickmann, 2022-10-26:
The problem is not due to the graphics system, it is due to the touch events being different.

If I fire up the **EventDemo** widget that is on the standard SD card content, click on the little square, drag it around and then let go, then I get the following events in Lua:

**2.7.1**:  `EVT_TOUCH_FIRST`, `EVT_TOUCH_SLIDE`, `EVT_TOUCH_BREAK`
**2.8rc3**: `EVT_TOUCH_FIRST`, `EVT_VIRTUAL_ENTER_LONG`, `EVT_TOUCH_SLIDE`, `EVT_TOUCH_TAP`

Is this change intentional, or by accident? If it is intentional, what is the rationale for the change?

pfeerick, 2022-10-26:
I wouldn't think it was intentional... `EVT_VIRTUAL_ENTER_LONG` for one looks out of place, and `EVT_TOUCH_TAP` also - almost as if it were incorrectly handled as a tap and a slide (as if not checking for touch AND release before firing the tap event?)?

jfrickmann, 2022-10-28:
So everything that I added in #703 has been anihilated, and even a slow touch is now a tap. 
Now I could of course dive head first into the code and try to add it back, but my feeling is that it would likely suffer the same fate as the previous code, so maybe better leave it to @raphaelcoeffic  to fix it?

jfrickmann, 2022-10-28:
I have fixed the tap events in #2637. This is not an elegant solution as it copies all of `internalTouchState` just to read `tapCount`, but it seems to work.
I have not tried to fix the issue with `EVT_VIRTUAL_ENTER_LONG` as I don't know if the GUI relies on that elsewhere - I will leave that up to @raphaelcoeffic!

raphaelcoeffic, 2022-10-28:
@jfrickmann so I did break it? 🤭 I really thought I had it right. What should I use best to try?

jfrickmann, 2022-10-28:
> @jfrickmann so I did break it? 🤭 I really thought I had it right. What should I use best to try?

Yes you did 🙄  But luckily, it should be easy to fix. You can use the Lua widgets that I have added to the standard SD-card content (EventDemo and LibGUI), and you can also create a new model from my SoarETX templates and play with the widgets they have, to test.

As for the `EVT_VIRTUAL_ENTER_LONG`, I assume that it is generated by this code:

https://github.com/EdgeTX/edgetx/blob/58742b526e093cfbfe118569a2e2f9a3bbd4032a/radio/src/lua/lua_widget.cpp#L46-L58

It seems to me that the `else if` part only gets reached if the event is not a key event (`LV_EVENT_KEY`)? What is the purpose of pushing `EVT_KEY_LONG(KEY_ENTER)` for a non-key event?

raphaelcoeffic, 2022-10-29:
@jfrickmann with LVGL, pressing on encoder wheel, or clicking on the screen (touch) **does not** generate `LV_EVENT_KEY`.

It uses the following events:
- `LV_EVENT_PRESSED`
- `LV_EVENT_RELEASED`
- `LV_EVENT_LONG_PRESSED`

In case you need to differentiate where the events are coming from, you can check the source like this (in the event handler itself):
```c++
lv_indev_type_t indev_type = lv_indev_get_type(lv_indev_get_act());
if(indev_type == LV_INDEV_TYPE_ENCODER) {
  // event from encoder wheel 
} else if (indev_type == LV_INDEV_TYPE_POINTER) {
  // event from touch sensor
}
```

raphaelcoeffic, 2022-10-29:
Regarding breaking the events: we moved to a **totally different UI framework**, with **100s of hours of work**. If something goes down the drain, I expect folks who use these features to raise the alarm and/or help fix them. It cannot possibly be my sole responsibility....

raphaelcoeffic, 2022-10-29:
@jfrickmann I added a version of your PR that does not rely on the internal state of the touch sensor driver. Indeed, as LVGL does not handle these gestures, we should not rely on API layer-bypass to have the feature. Instead, we should re-implement it based on what LVGL native events.

Could you please re-test if that works better for you? Is anything else missing?

jfrickmann, 2022-10-29:
@raphaelcoeffic , I was aware that you were moving everything to the LVGL framework, and I do believe that I am raising the alarm here, am I not? If you want me to test something at an earlier stage, then please ping me when you think that it is ready for testing, because I cannot keep track of all of your progress.

I certainly do appreciate the impressive amount of work that y'all are pouring into this project, and when it starts to stabilize some day, it will be a great radio OS 😎 

As for the TAP events, there was actually a fair amount of discussion going on here, back when I first implemented events in Lua - please see #422 to refresh your memory. To me, this is not a sacred thing, and I can adapt if it is changed to something else, as long as it is sensible. And if everything else has been moved to LVGL, then it makes sense to me that Lua also follows along. But the changes need to be clearly documented!

If the internal tap state is no longer used, then maybe you also want to remove that code from tp_gt911.h and tp_gt911.cpp?

I built and tested your PR in the simulator, and it seems to do the same as my PR, so good!  But it still pushes `EVT_ENTER_LONG` when I put a finger on the screen and don't move it for a second or so. Is that intentional? Otherwise, it should also be fixed.

Another issue that I noticed, is that the popup menu for selecting a new Lua widget suddenly jumps to the end when I have scrolled ~half way down with the button. So I could only select the EventDemo widget with touch, as the scroll button would skip over it.

gagarinlg, 2022-10-29:
> If the internal tap state is no longer used, then maybe you also want to remove that code from tp_gt911.h and tp_gt911.cpp?

I started a complete rework of the touch drivers a while ago, to decouple touch controller drivers from specific radios and start having a HW driver lib, that is used by the radios, depending on what hardware they have. NV14/EL18 and PL18 share alot of the hardware, but have their own copies of the drivers, which is a bad thing.
The design of the touch stuff, before LVGL, was not good. UI code wrote to a structure also used by the HW driver and other strange design decisions. Since LVGL only expected the touch drivers to emit touch changed events, with coordinates and if it it still touched or not, all other stuff will be removed from the drivers and, in my opinion, never belonged into the drivers. 
We should try to emulate the old behavior as good as possible, to make the transition easier. In the long run I would like to expose more of LVGL to the lua scripts, so that all LVGL drawing optimizations, with partial screen redraws, also work with lua.
But this a relatively large amount of work, where you and @JimB40 are needed for input before starting the work and later for testing and feedback and it will break compatibility with OpenTX completly.

I could start a seperate discussion for the great UI interface change, if you are interested. But this is a long term project. It will not be finished for 2.9

raphaelcoeffic, 2022-10-29:
@raphaelcoeffic the useless parts will be removed, as @gagarinlg pointed out.

Regarding long press: should we or should we not have them? How was it before? I have no problem restoring the previous state if nobody is using these events and it would otherwise confuse existing scripts.

jfrickmann, 2022-10-29:
I agree that the old design was messy, and I am personally completely on board with using standard LVGL events also in Lua. I implemented the touch interface that we had reached consensus for after a community discussion. For my own work, I do not expect that changing it will to be a big problem, as it is based on the Lua widget LibGUI that provides a common GUI library with touch support in Lua.

If I remember correctly, I added the tap count to the touch state in the driver, because the Lua widget part sometimes missed events due to the relatively low run frequency. But all of that would be irrelevant if we limit ourselves to using the standard set of events from the lower layers.

gagarinlg, 2022-10-29:
The event emulation should be done low enough, that it does not depend on the LUA cycle time.

pfeerick, 2022-10-31:
> Another issue that I noticed, is that the popup menu for selecting a new Lua widget suddenly jumps to the end when I have scrolled ~half way down with the button. So I could only select the EventDemo widget with touch, as the scroll button would skip over it.

Try scrolling a little slower - you probably scrolled just a little too fast and triggered the rotary encoder acceleration, hence the jump.

PR #2646 (2022-10-31): fix(lua): re-implement LUA "tap" events
As LVGL does not support natively the gestures handled by LUA with respect to touch sensors, this PR re-implements the "tap count" instead of relying on the internal driver exporting the data.

Fixes #2617
commit 6763c979 (2022-10-31): fix(lua): re-implement LUA "tap" events (#2646)

Fixes #2617
commit 93a007cf (2022-11-02): fix(lua): re-implement LUA "tap" events (#2646)

Fixes #2617
closed 2022-10-31 by pfeerick