EdgeTX/edgetx #2619: Problem navigating through the SD Card content with rotary encoder

ParkerEde, 2022-10-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I go to the system menu and open the SD card browser, there is a problem with the highlight bar. 
For example, when I open the fourth folder in the list, the folder opens and the highlight bar is on the fourth entry in that folder. When I open the sixth folder, the highlight bar is on the sixth entry in that folder. And so on...

Same behavior is in Companion simulator

### Expected Behavior

After entering a folder, the highlight bar should be on the first entry within the opened folder.

### Steps To Reproduce

1. open SD Card browser
2. Scroll to the fourth Folder with rotary encoder
3. press Enter
4. see the highlight bar is on the fourth entry (count includes [..])

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild commit d7598c8ed8586f

eshifri, 2022-10-22:
Similarly, when going to the upper level folder, selection moves to the first item in the list.

raphaelcoeffic, 2022-10-31:
Emptying the table should trigger invalidation of the selected item, which is probably missing right now. @eshifri in case you wanna have a look ;-)

raphaelcoeffic, 2022-10-31:
Ok, in fact, it **is** reset, but for some reason the table does not pick the change (whereby `lv_obj_invalidate` is called properly).

raphaelcoeffic, 2022-10-31:
I have it tracked down, but  the fix is not easy to find.... The issue is this little bugger:

https://github.com/EdgeTX/libopenui/blob/f8f63688ea12dd55c582d835dee5848ac2c1bfcc/src/widgets/edgetx_table.c#L30-L48

It effects that the line selected in `LV_EVENT_VALUE_CHANGED`, which is sent from `LV_EVENT_RELEASED` gets overwritten with the value set in the handler for `LV_EVENT_PRESSED`....

There is no easy solution for that.

closed 2022-10-31 by pfeerick