EdgeTX/edgetx #2621: Telemetry widgets setup screen and Page> & Page< navigation

jurgelenas, 2022-10-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When inside the telemetry widget setup for each screen:

[image]

I can navigate through different screens, but it does retain the same widget outlines from the previous screens.

### Expected Behavior

Navigation using Page> and Page< buttons when in widget setup mode should work as expected. 

### Steps To Reproduce

-

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

I am using 2.8 RC3 EdgeTX version.

eshifri, 2022-10-23:
Can you please describe the sequence? This does not look like a setup screen.

jurgelenas, 2022-10-23:
> Can you please describe the sequence? This does not look like a setup screen.

1. Go to telemetry menu by pressing "TELE" button
2. Click "Setup widgets" button in any view (except for "top bar" probably)
3. Try navigating using PAGE> or PAGE< buttons when you have multiple views in place

eshifri, 2022-10-23:
I did not know these buttons work in the widget setup screen :-), but they do work for me.
Can you please give it another try on the new model?

pfeerick, 2022-10-24:
> I did not know these buttons work in the widget setup screen :-)

I don't think they are supposed to... IMO it doesn't make sense to be able to page between widget screens when you explicitly pressed setup widgets for a specific view. 2.7.1 doesn't behave this way. But I'm not against the idea if that is what people want.  :grin: I setup two views with different layouts, and while `PAGE>`/`PAGE<` did do something, it wasn't right (basically the same as what Jurg showed - outline for selected view stacked ontop of the second view content) 😆

PR #2630 (2022-10-26): fix(color): Disable invalid PG_UP and PG_DN handling on widget setup screen

Fixes #2621

Summary of changes: Disable PAG_UP and PG_DN buttons on the widgey setup scree.
The buttons were working incorrectly and there is a consensus (I feel) that they should not have any effect there.
closed 2022-10-26 by pfeerick