EdgeTX/edgetx #2624: edgeTX companion wrong channel simulator in jumper t-lite

zakopanypouszy, 2022-10-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Hello, I recently switched from openTX to edgeTX on my t-lite jumper and it all went fine, it works fine as it should be on openTX. The problem arose when using companionTX because the simulator shows incorrect channel movements on both the existing models and those newly created in the companion. On the radio, by default (mode2), I have to close channel 3 to the throttle stick, but the open tx simulator shows the movement of channel 2. The remaining inputs and channels are compatible with the mixer.

### Expected Behavior

correct simulator working

### Steps To Reproduce

simulation

### Version

2.7.1

### Transmitter

Jumper T-Lite

### Anything else?

[image]

pfeerick, 2022-10-24:
Can you upload a `.etx` file with a model that exhibits that behaviour (just change the extension / add `.zip` to the end, and drag it to the text view area to upload). I can't reproduce that behaviour here... correct channel is moving for throttle, and simulator radio is saying correct input is moving :-/

zakopanypouszy, 2022-10-25:
[JUMPEREK EDGETX.zip](https://github.com/EdgeTX/edgetx/files/9862936/JUMPEREK.EDGETX.zip)

pfeerick, 2022-10-28:
Can you try uploading this again as the file seems to be corrupt.

zakopanypouszy, 2022-11-05:
[jumperek.zip](https://github.com/EdgeTX/edgetx/files/9944707/jumperek.zip)

sorry it took so long but I didn't have access to a computer with a radio, now it's an etx file compressed to zip format. The file includes all my models from the radio, however, checking individual models during the simulation, the problem of the wrong throttle channel occurs both on the quad-expresslrs and airplane-flysky ia6b models