EdgeTX/edgetx #2625: Trim on throttle (mode 2), CH3 got at minimum value on each restart.

macfly1202, 2022-10-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When power on the radio, the trim for throttle is going full down even if corrected on previous session. 
I made a video to describe the issue:
https://youtube.com/shorts/mB6EDoxKrS0?feature=share

This model has been converted from OpenTX 2.3 for your information.
Trim is confuigure on radio as "extended" trim and "exponential" step. 

### Expected Behavior

The trim position must be record and get same position after radio restart.
 

### Steps To Reproduce

Power on radio.
change trim throttle.
power off radio 
power on radio.
See the trim throttl evalue go full down. 

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2022-10-24:
What's causing the "stuck key warning"? Is the throttle-down trim stuck down by any chance?

With both 2.7.1 and 2.8.0-rc3 and a model configured with expo trim, extended trim, the throttle trim is staying where it is set. If I hold the throttle trim down on powerup, I get the stuck key warning (without indication of which key is stuck) and throttle trim will shoot to the bottom, exactly like yours is doing when it turns on... you can actually see the trim going down just after the warning screen is cleared.

gagarinlg, 2023-05-05:
probably a HW-issue

closed 2023-05-05 by gagarinlg