EdgeTX/edgetx #2629: Double tap on wigdet doesn’t work

JimB40, 2022-10-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Double tapping on widget does nothing

### Expected Behavior

It should switch widget to full screen.

### Steps To Reproduce

Double tap on widget placed on main view

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

ETX 2.8 RC3

pfeerick, 2022-10-25:
No, but I thought the same at first. This was a deliberate change. Press and hold for the menu to get the fullscreen option. The same as a long press of the ENT/roller is now also needed to prevent inadvertently entering widget selection mode for non-touch. 

@raphaelcoeffic Anything more?

raphaelcoeffic, 2022-10-31:
I was think on adding some generic gesture support in 2.9, but for now, this is how it works, yes.

JimB40, 2022-11-01:
You mean double-tap on widget swiching mode to 'Full screen" with key capture in was switched off in 2.8???

BTW I remember this discusion about naming.
So what should I display to user now 
[image]
1. "Go Full Screen to activate keys"? - confusing as we already are in full screen
2. " Tap and hold to get menu, then select Full Screen option to activate keys"? - Sounds like manual

- Full Screen - is display layout
- App Mode - is state when LUA widget gets key inputs

with non-full-secren layout both are enabled but with full-screen layout it's hard to make is intuitive and user frendly.

UX with widgets that ocuppies Full Screen became very confusing now. I can see it by user questions/issues staring with "how the hell do I ... "

raphaelcoeffic, 2022-11-01:
> “Go Full Screen to activate keys"? - confusing as we already are in full screen

the confusing part is calling it fullscreen in the first place. If you capture the keys, you cannot switch screens anymore. So it feels weird when having different screens. If one of them is the kind of full screen you mean, then keys are captured at once and you cannot switch to the next screen. Same with sliding to the next screen. That does not work. So I’d prefer to quite calling that full screen, and quit making it full. Basically adding some kind of mandatory borders with something in it when not in the real full screen mode.

raphaelcoeffic, 2022-11-01:
To make it clearer, and with your terms:
- "Full Screen" - the display layout - **should not exist** at all, as it breaks other things (like triggering main menu with touch).
- "App Mode": should be the only full screen, as it enables to have the things you want to use.

JimB40, 2022-11-02:
Well for Apps like Yappu, iNav or my Toolbox or seen somwhere log anaylser, full-screen layout is way to go.
As we need as much UI space as possible. Another thing is how to implement it in best intuitive way.
Most asked question is  "how do I setup it to auto start and be running full screen with radio power-up"

gagarinlg, 2022-11-02:
How about a way for a Lua widget to put itself into full screen mode and out of it?

JimB40, 2022-11-02:
Yes I was thinking about lua function complementary to lcd.exitFullScreen() -> lcd.enterFullScreen().
But stucked with caseL
1. You open main view with full-screen layout so script detects no key events and auto-switch  to App Mode
2. You press EXIT and script knows you want to leave App Mode and goes out of it.
3. Now what? :) Script wants to execute condition 1 and go back to App Mode or user has to go back to App Mode manualy (as it is now)

We may (free thinking)
1. hard code PAGE_NEXT_LONG & and PAGE_PREV_LONG to switch between main views (making those events unavailabe to LUA script). But then on NV14/EL14 all key events available via Joysticks in system & way to switch between key capture and trim dunction
2. Use swipe left/right to switch main view. Swipe left/right then will not be available to LUA only for OS and swiping will change main view exiting App Mode if widget was in this state

Surely something to be tested on hardware.

offer-shmuely, 2023-04-03:
closing the duplicated issue, moving the comments to here as a reminder that "we like this feature" :-)

As we do like to **_promote_** smarter widgets with **_app-mode awareness_** .
For this we need to make the entering/existing app-mode to be as simple & fast as we can.
This will encourage users to use it.

I hope there will be a way to add back the double-tap as it works in 2.7
This in regardless of the current menu that can be stay (maybe with renaming to "open as app")

I already added double-tap as exit from app-mode to some of the widgets
[image]

https://github.com/EdgeTX/edgetx-sdcard/blob/master/sdcard/c480x272/WIDGETS/GaugeRotary/main.lua#L252