EdgeTX/edgetx #2639: 2.8 RC3 issue: the SYS button does not allow access to "main" section

Billy60, 2022-10-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The menu for "main" cannot be reached from the SYS left hand button.
The Tx displays a message "script syntax error  luaLoadScripts(standa): no run function"

### Expected Behavior

The "main" section of the sys menu would be reachable.

### Steps To Reproduce

1 start the Tx
2 after the welcome messages, press on SYS
3 get to the Menu. Look for "main"
4 get the error message above.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Edge 2.8 RC3

pfeerick, 2022-10-28:
Just to confirm, this is at the "Tools" screen? What is "main"? This sounds
like a script that you have installed either in the wrong place (i.e. Maybe
a widget or telemetry that you have installed into SCRIPTS/TOOLS instead of
WIDGETS) or is otherwise corrupt or invalid.

Billy60, 2022-10-29:
No, this is when you press on the SYS button. The very next menu which appears has items like 'external rf' internal rf', and one called 'main'. This one allows the adjustment of the sound, buzzer, luminosity, hour, battery etc.
This is not a widget but is from the Fw directly.

raphaelcoeffic, 2022-10-29:
@Billy60 what you are looking for is now behind sub-menus accessible with the buttons. This avoid long screens with lots of scrolling.

raphaelcoeffic, 2022-10-29:
@Billy60 regarding the error message `script syntax error luaLoadScripts(standa): no run function`, I believe you have some LUA script installed at the wrong place. The "Tools" screen is trying to scan for these scripts and detects a script that is not a "Tool", but is in that directory.

Billy60, 2022-10-29:
Raphaelcoeffic, this is about a fw feature, not a external script.
[image]

This is a photo of the menu from the SYS button. The page is in TOOLS which pops in automatically.
Now if this sub menu is in the wrong place, where should it  be ?
Thanks !

Billy60, 2022-10-29:
Sorry Raphael, I got it.
In fact this 'main' item is indeed from some external script and not the one I was looking for, which is in the next 'tool' item further down on the same page.
I got confused because of these two having the same name 'tools'.
Sorry again, keep up the fantastic work !

rotorman, 2022-10-29:
@Billy60 The issue can be closed?

Billy60, 2022-10-29:
Yes, I suppose so.
Thanks to all !

closed 2022-10-29 by raphaelcoeffic