EdgeTX/edgetx #2657: Trainer Signal Lost between TX16sMK2 & Zorro - Throttle goes above 100%

nrlourenzo, 2022-10-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I'm trying to bind a TX16s MK2 (Master) with a Zorro (Slave) via 3.5mm cable but I'm facing some issues.

All works perfectly, however, the throttle on the Zorro seems to go above (& below) the 100% of the TX16 and when that happens I get a Trainer Signal Lost message. When it comes below the 100% I get a Signal Recovered

The workaround I found is to set the Throttle Mixer to 97 on the Tx16, but I'm sure there's a better way.

Both TX are using the EdgeTx 2.7.1, trims are centered, both were calibrated many times and I've also calibrated the Tx16 Trainer page where the throttle goes up to 102.4%. 

https://photos.app.goo.gl/YK9F1SRWu5TByyFRA

I can send a better quality video later if you guys need.

Thanks

### Expected Behavior

Stop losing Master/slave connectivity

### Steps To Reproduce

On Zorro TX:
Push throttle up to 100%
Push throttle down to -100%

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

video available here: https://photos.app.goo.gl/YK9F1SRWu5TByyFRA

raphaelcoeffic, 2022-10-31:
@nrlourenzo you should check the "Limits" screen on the Zorro.

We do have some validity checks on the trainer input: the pulse length for each channel must be between 800uS and 2200uS. If the Zorro is outputting more than that, this is what you will see.

raphaelcoeffic, 2022-10-31:
If you can read some code, you will see that the trainer settings (factor / multiplier) is only applied **after** that check:
```c++
      if (val > 800 && val < 2200) {
        ppmInputValidityTimer = PPM_IN_VALID_TIMEOUT;
        ppmInput[channelNumber++] =
          // +-500 != 512, but close enough.
          (int16_t)(val - 1500) * (g_eeGeneral.PPM_Multiplier+10) / 10;
      }
      else {
        channelNumber = -1; // not triggered
      }
```

> workaround I found is to set the Throttle Mixer to 97 on the Tx16

Also, it is worth noting that changing anything on the `Master` side should not have any influence.

nrlourenzo, 2022-11-02:
Thanks Raphael.

The limits are the default on both 988-2012 for all the channels. And I believe the configuration is alright.. Today I noticed that it's not just on Throttle (ch3) that it happens..

Please check these videos, will show it better and maybe you'll see something I cannot:

https://photos.app.goo.gl/aS6btPsYEhpyHajs8

If there's a way to export a debug or config let me know. Thanks again.

nrlourenzo, 2022-11-05:
@raphaelcoeffic

I found the issue.. on Model setup on Zorro, under Trainer, if the Channel range is above 10 this happens and it was set to 12.. I lowered to 10 and issue solved.

Video showing this:  https://photos.app.goo.gl/5cvqR69GpuR3v9Rf7
What's the explanation? Or is this a bug?

Cheers

rotorman, 2022-11-05:
With 22.5 ms frame, you probably just cannot cram more channels in there, before the reset pulse, that is used for channel 1 detection, gets too short. What is really going on could be best observed with a logic analyzer or a scope.

raphaelcoeffic, 2022-11-05:
@nrlourenzo Thx! That means we need to constraint the max number of channels properly, you should not be able to do this.

raphaelcoeffic, 2022-11-06:
After some research, it seems we should actually force a longer frame in the driver when the number of channels and/or channels values require it.

However, we already have a PR for `2.9` (see #1750) which basically overhauls the complete trainer drivers and make them much more solid, so that I'd like delay any changes and consider this a *known issue*.

nrlourenzo, 2022-11-07:
@raphaelcoeffic @pfeerick @rotorman 

Thanks to all for your time.

rotorman, 2022-11-08:
To visualize the issue, here the trainer PPM signal output with 22.5 ms PPM frame and 300µs negative pulse with 11 channels highlighting the reset pulse width:
[image]

With 12 channels, we easily go (in channel center positions already) under 5 ms reset pulse width:
[image]

raphaelcoeffic, 2022-12-10:
@rotorman would you mind re-testing with #1750? I think if there is anything left, we should fix it as part of that PR.

rotorman, 2022-12-10:
As #1750 is already merged, I tested with main 78f75c5 on TX16S mkII.

Unfortunately this needs more work!

When trainer is set as Slave/Jack, channel Range e.g. to CH1 to CH11 and PPM frame set to 22.5ms, 300µs negative pulse and - polarity, this is what I get (the output depends on the channel values) in worst case:

[image]

Reset pulse way too short.

Another observations with 16 channels:

[image]

Here it does not run into reset pulse, but also in the user interface, it does not automatically adjust the PPM frame length to match the true value.

gagarinlg, 2023-05-06:
Looks like we have to limit the number of channels a slave can transmit. Does anyone else have a better idea?

phileville, 2023-05-19:
I noticed that this in taggd with "Documentation".  What is the final solution here - do we make a SW change to limit the number of channels or shall I just add it to the User guide that only Channels 1-10 can be confingured?

pfeerick, 2023-05-19:
https://github.com/EdgeTX/edgetx/issues/2657#issuecomment-1304729979 was the reason it was 'Documented' - it's been marked as a known issue in the release notes since probably 2.8.0

> Defining too many channels for trainer link will result in intermittent trainer signal. Either increase the PPM frame length or decrease the number of channels. https://github.com/EdgeTX/edgetx/issues/2657

So for now, user guide probably just needs to say something similar - i.e. if you want more than 10 channels, you need to ensure the frame length is longer so they all fit. Until some UI change is made to adjust the frame size or limit the max channels based on the frame length?

pfeerick, 2023-09-20:
#3622 implements the aforementioned UI adjustments, so this issue should be generally resolved now.

closed 2023-09-20 by pfeerick