EdgeTX/edgetx #2664: Event handling in lua is not consistent

lshems, 2022-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Lua issue.

Key events are key (pun intended) in controlling scripts with user interactions.

A big improvement was made with the introduction of virtual events, allowing scripts to interact correct on multiple radios.

However, there are some inconsistencies:

1) long events should be triggered before break events. This is the case for most events, but not for the long exit, which comes after the break exit event (EVT_EXIT_BREAK). normal' virtual events should correspond to the BREAK of the key pressed.
This is especially important for the exit event, because a long exit is used to kill a script, and should not trigger a last exit_break before quittting, causing unexpected behaviour.
2) not all  events have a constant associated. EVT_EXIT_LONG doesn't have one.
3) not all events are triggered, especially when virtual events already exist linked to the same button.
4) the virtual next page is triggered on long press page, after the previous page event. This seems wrong.

### Expected Behavior

Correct and consistent key events

### Steps To Reproduce

Just trigger events and show in debug using print(event)

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T16

lshems, 2022-11-01:
Super.

Exit long could be passed with reason of you would allow the run cycle to finish before killing the script.

This would allow cleanup before killing.

How about VIRTUAL_FORCE_EXIT.

As for the other events, why not consider creating full abstraction of the physical events and only keep the virtual ones?

JimB has a script and created a library for all events. @JimB40 could you provide more info?

lshems, 2022-11-01:
Also, if the https://github.com/EdgeTX/edgetx/pull/2573 should fix the exit thing I assume this is not in RC3 yet?

gagarinlg, 2022-11-01:
> Also, if the #2573 should fix the exit thing I assume this is not in RC3 yet?

https://github.com/EdgeTX/edgetx/releases
it is not in RC3.
When you want to test it, use the nightlies.

lshems, 2022-11-01:
Ok. Is it planned for 2.8.0?

raphaelcoeffic, 2022-11-01:
> Ok. Is it planned for 2.8.0?

We cherry-pick all fixes after merging to `main`.

raphaelcoeffic, 2022-11-02:
> you would allow the run cycle to finish before killing the script

**kill** is **kill**. Proper **exit** should be implemented in a usual way, so the user does not have to kill the script.

> fix the exit thing I assume this is not in RC3 yet?

No, either next RC, or nightly.

> As for the other events, why not consider creating full abstraction of the physical events and only keep the virtual ones?

Virtual events are just aliases to the "normal" ones. What you are asking for is filtering, right? So basically only emit events to LUA that have a constant defined, is that right?

raphaelcoeffic, 2022-11-02:
> the virtual next page is triggered on long press page, after the previous page event. This seems wrong.

See #2668

lshems, 2022-11-14:
1) long events should be triggered before break events. This is the case for most events, but not for the long exit, which comes after the break exit event (EVT_EXIT_BREAK). 'normal' virtual events should correspond to the BREAK of the key pressed.
This is especially important for the exit event, because a long exit is used to kill a script, and should not trigger a last exit_break before quittting, causing unexpected behaviour.
FIXED IN 2.8 RC4

2) not all events have a constant associated. EVT_EXIT_LONG doesn't have one.
Better to move requirements to #2736 as all are listed there.

3) not all events are triggered, especially when virtual events already exist linked to the same button.
Better to move requirements to #2736 as all are listed there.

4) the virtual next page is triggered on long press page, after the previous page event. This seems wrong.
FIXED IN 2.8 RC4