EdgeTX/edgetx #2665: Difference at Telemetry ratio/instance between radio and companion

elekSENDER, 2022-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In Model Settings/Telemetry parameter Ratio values are displayed in the radio 10x bigger than in companion.
The value Instance is +1 displayed in the radio than in companion.

The displayed values in the radio of the active measured sensors are still correct in comparison to OTX.

[image]
[image]
[image]
[image]

[image]
[image]

### Expected Behavior

same values in companion and radios

### Steps To Reproduce

read/write companion/radio by changing parameter Ratio/Instance in telemetry

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

V2.8.0-RC3

this is a substitute for [#1650](https://github.com/EdgeTX/edgetx/issues/1650)

frankiearzu, 2023-05-23:
Just FYI,
I did fix the ESC VBEC since it actually needed to be multiplied by 5 to get a decent value.
For example, raw was 121, so 121*5=605, that with 2 decimal place is 6.05 that is the correct value. (and matches the Spektrum documentation, and testing with an Spektrum TX).

elekSENDER, 2023-05-24:
> Fixes: ESC VBEC since it actually needed to be multiplied by 5 to get a decent value.
> For example, raw was 121, so 121*5=605, that with 2 decimal place is 6.05 that is the correct value. (and matches the Spektrum documentation, and testing with an Spektrum TX).

Great to hear that we don´t need this workaround anymore for correct Spektrum telemetry data.
I tried this with a YGE-ESC and it worked very well.

> This might have been solved by: #3615
> The value is correctly converted by the firmware, instead of trying to fix it afterwards using ratio.

Yes, for Spektrum. But this issue is not solved in general, because i use ratio as well for FrSky telemetry sensors and the ratio difference between radio and companion still exists there.

aeropic, 2023-07-10:
> 
> 
> Radiomaster TX16S / TX16SMK2
reproduced in TX16S MKII under EDGETX 2.8.4

closed 2026-09-21 by github-actions[bot]