EdgeTX/edgetx #2669: Scrolling in contextmenu "inputs" and "mixes"

ParkerEde, 2022-11-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

in the last nightly it is no longer possible to select all items in the context menus of the "Inputs" and the "Mixes". "Move..." and "Copy...." are completely skipped.

### Expected Behavior

each context entry should be selectable

### Steps To Reproduce

1. open Companion or firmware Simulator
2. Enter Inputs or Mixes
3. Edit line
4. try to scroll to "copy line"
5. you'll see it's not possible anymore

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.8.0 selfbuild 8ce3612

eshifri, 2022-11-03:
Cannot reproduce it.
In the simulator mouse scroller moves cursor in the menu by two lines ( a known problem ?).
But I can get to Copy by moving from bottom up.

ParkerEde, 2022-11-03:
Hi @eshifri 
Did you try it in Companion Simulator or on radio? It seems to be an only companion issue.

ParkerEde, 2022-11-03:
https://user-images.githubusercontent.com/15068446/199673928-dcd027da-afb5-4994-bc2e-712da38530c1.mp4

ParkerEde, 2022-11-03:
It's the same in context of model select menu:

https://user-images.githubusercontent.com/15068446/199677169-eab4fc41-858d-4acc-8d85-5b36b15d012a.mp4

ParkerEde, 2022-11-03:
@eshifri 
> In the simulator mouse scroller moves cursor in the menu by two lines ( a known problem ?).

This bug was already fixed in https://github.com/EdgeTX/edgetx/pull/2319

eshifri, 2022-11-03:
I'm using the latest code (self-compiled) and this bug was never fixed for me.
I tried it in the stand-alone simulator (launching without CP), but this must be the same.

ParkerEde, 2022-11-03:
However, this new problem is only in the context or drop-down lists. In tables or other areas such as the SD card browser, scrolling is normal in the simulator.
BTW I could also reproduce the problem with the TX16S in the simulator.

pfeerick, 2022-11-04:
Ok, just repo'ed this on the current 2.8 head companion build, but not with rc3, so that at least narrows down the possible culprits. 
TX16S

https://user-images.githubusercontent.com/5500713/199893884-a7849748-91e2-4906-8306-102a5fc96daa.mp4

raphaelcoeffic, 2022-11-04:
May I ask which controls you use to move the simulated roller? Is that the mouse wheel, or is that `+/-`? (or both?)

pfeerick, 2022-11-04:
Scroll wheel.

eshifri, 2022-11-04:
Oh, it was working for me this way from the very start of 2.8. I thought it was the behavior described in #2319 :-)
After playing with it today, I can add that 
1. the size of the jump depends on the mouse used. 
2. The jump is not repeatable: sometimes I see it going from the last line to the top; sometimes the step is two lines.
3. I can get to Copy by the following sequence: Edit->Insert after->Delete->Copy. :-)

raphaelcoeffic, 2022-11-04:
@eshifri the latest rotary encoder acceleration change broke it. It relies on the driver to compute the time diff between events, which the simu wasn't doing (neither plain `simu` nor Companion).

ParkerEde, 2022-11-04:
> May I ask which controls you use to move the simulated roller? Is that the mouse wheel, or is that `+/-`? (or both?)

it happens with both, mousewheel and "+/-"

pfeerick, 2022-11-04:
See how you go with https://github.com/EdgeTX/edgetx/pull/2676 @ParkerEde ... you might get to it before me ;)

ParkerEde, 2022-11-04:
I've checked out origin/fix-simu-encoder and build only libedgetx-x10express-simulator.dll. 
Works as it should - thx :-)

ParkerEde, 2022-11-04:
libedgetx-tx16s-simulator.dll is fine too ;)

PR #2676 (2022-11-05): fix(simu): rotencDt must be updated as well
The acceleration calculation is relying on it to compute dx/dt.

Fixes #2669
closed 2022-11-05 by pfeerick