EdgeTX/edgetx #2671: Drive name of EL18 display NV14 when connecting USB to computer

richardclli, 2022-11-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Drive name shows "NV14" when connecting USB to computer.

### Expected Behavior

Drive name should shows "EL18" when connecting USB to computer.

### Steps To Reproduce

1. Switch on EL18
2. Connect USB to computer
3. In explorer it shown NV14

### Version

Nightly (Please give date/commit below)

### Transmitter

Flysky EL18

richardclli, 2022-11-03:
Seems already fixed in 2.8 branch.

closed 2022-11-03 by richardclli

nightflasher, 2022-12-01:
well...NV14 now shows up as EL18 in debian/ubuntu.

gagarinlg, 2022-12-01:
This is a hard coded text somewhere. It needs a better, more flexible solution

pfeerick, 2022-12-02:
If it's coming up as EL18 - it has probably been flashed with the EL18 firmware, not the NV14 firmware... as it is hard-coded here - so cannot change dynamically. You should be able to check on the radio settings-version page - it will say it's the el18 variant instead of the nv14 variant. Other than cosmetics like that, there is no functional difference between the builds. 

It does need to be more flexible - preferably runtime - but it was good enough for 2.8. 

https://github.com/EdgeTX/edgetx/blob/ac16299788c88194ecb669064a3779581181c372/radio/src/targets/nv14/board.h#L480-L488